Plutoberth/SonyHeadphonesClient #18: Option to turn off NC and ambient sound

quantumfrost, 2020-10-05:
Really like the program, but I don't see any option to turn off both NC and ambient sound. It's useful if you want to save battery on the headphones. In the Android app, you can do this by turning off the "Ambient Sound Control" toggle.

[image]

Plutoberth, 2020-10-05:
This is definitely possible. I remember doing this accidentally during testing, but I don't remember how I played with the parameters to make it happen (should be fairly trivial). I'm pretty busy now though, so it'll probably take a while until this is implemented if no one else takes it.

guilhermealbm, 2021-03-09:
I'm working on it!

PR #36 (2021-03-14): Add option to turn off NC and ambient sound
This fixes #18. After taking a look over bluetooth messages, found out that ambient sound control is turned off when sending vol=-1.

The only problem I faced is when checking and, after, unchecking the box. It returns a "Invalid checksum!". Same happens when do it at "Focus on Voice" checkbox, so I think that's maybe a problem with XM4's (and not my changes), as other people said in #27.

I tested it in XM4's and Windows. Hope it helps! :)
closed 2021-03-14 by Plutoberth