Plutoberth/SonyHeadphonesClient #20: How can I reverse engineer my headphone to feature to the app?

ViniDalvino, 2020-11-02:


Plutoberth, 2020-11-02:
Hi,

If you're interested I can share the research repository with you. It's basically a decompiled version of the app with many comments and names (they ran ProGuard on the app, so almost all names have been stripped).

miaachan, 2020-11-04:
> 
> 
> Hi,
> 
> If you're interested I can share the research repository with you. It's basically a decompiled version of the app with many comments and names (they ran ProGuard on the app, so almost all names have been stripped).

I am also interested!

ViniDalvino, 2020-11-04:
I am interested. @Plutoberth

BenConstable9, 2020-12-28:
I am interested as well if this is available?

Plutoberth, 2021-01-02:
After further consideration I've decided not to test Sony's legal team. If you want, you can do this yourself - I just decompiled it using jd-gui, and then exported the text files for easier work in IntelliJ. The protocol is documented as code in this repository. Additional commands would be implemented similarly to the function [serializeNcAndAsmSetting](https://github.com/Plutoberth/SonyHeadphonesClient/blob/master/Client/CommandSerializer.cpp#L174), for example.

closed 2021-01-02 by Plutoberth