Plutoberth/SonyHeadphonesClient #22: [Feature Request] Volume Control

Asheboy, 2021-01-04:
The Sony Wf-1000xm3 do not support volume control on Linux. It would be great if the Bluetooth commands for it could be reverse engineered.

Plutoberth, 2021-01-05:
I don't know much about the actual Bluetooth HID protocol, but it's safe to assume that there's nothing headset-specific about volume controls. Another example of a feature that's not headset-specific is battery readings. This tool provides support for commands that usually require the Sony app to use - like setting noise cancelling controls. 

It seems weird to me that volume controls don't work for your headphones on Linux, btw. Is this issue well known?

semvis123, 2021-01-05:
I want to add that this app is currently not even supported for Linux, but if you could contribute to this project if you want.
And I'm not sure if the wF-1000-xm3 will work with this client.

Asheboy, 2021-01-06:
@Plutoberth Turns out this is a well known issue, with support being added to the most common audio Linux sound system soon. 

Also, it does turn out that volume is standardised, anyone looking for Linux information may find this helpful https://stackoverflow.com/questions/28191350/is-there-any-way-to-control-connected-bluetooth-device-volume-in-linux-using-com.

closed 2021-01-06 by Asheboy