Plutoberth/SonyHeadphonesClient #25: Minimized 100% CPU

Myp3a, 2021-01-21:
**What happens?**
When the app is minimized, it consumes entire CPU core. Looks like some UI check without a timer - when in foreground or overlayed by another window, everything is fine

**How to reproduce**
1. Start app
2. Minimize it
3. Observe one core fully occupied in Task Manager

**What should happen**
Minimizing shouldn't affect CPU usage

**Version**
Headphones Client: v.1.0.2, Aug 20 Release
OS: Windows 10 20H2, latest stable

Plutoberth, 2021-02-07:
I see. This is probably caused one way or another by imGUI. It tends to cause things like this since it constantly re-renders the window. I'll check it out and report

Plutoberth, 2021-02-11:
This was caused by DX11 not vsyncing when the window is minimized. It has no buffer to write to, so it just gives up and doesn't wait. You can either build from master in a few minutes when I push the fix, or wait a bit for the next release.

See https://github.com/ocornut/imgui/issues/2496

PR #30 (2021-02-11): fix minimized cpu usage error
Closes #25
closed 2021-02-11 by Plutoberth