Plutoberth/SonyHeadphonesClient #26: update advice

Eleutherios0, 2021-01-31:
The program works very well, it would be nice if you add a shortcut with the new update

Plutoberth, 2021-02-07:
Do you mean a desktop shortcut? You can just place the program wherever you'd like, and then `Right click` --> `Send to` --> `Desktop`.

Eleutherios0, 2021-02-07:
No I didn't mean that.  For example, pressing the num1 key while I'm in the game and going into ambient mode

Plutoberth, 2021-02-07:
Ahhhhh got it. I'm not sure how I can capture keys globally (I saw something about it in imGUI, but I think that it's not mature) and there are more important issues, so I'm not going to implement it (at least in the near future).

Eleutherios0, 2021-02-07:
Thank you for making this application

semvis123, 2021-02-07:
System tray functionality could be added to get similar experience as the macOS, but this can not be controlled by keyboard shortcuts.

For example:
https://github.com/zserge/tray

Plutoberth, 2023-04-26:
not planning to do this, sorry :(

closed 2023-04-26 by Plutoberth