Plutoberth/SonyHeadphonesClient #27: WH-1000XM4 support

templeman, 2021-02-07:
Opening this issue to gauge interest for WH-1000XM4 support. I own this model and would love to be able to use this client with it. I have zero experience in C++, but I'm happy to test and provide logs for anyone who might choose to tackle this. I should be able to run either the Mac or Windows version (via Bootcamp).

FWIW it seems the app in its current state is close to being able to support the XM4s. Here are my results when attempting to use it with my pair:
_Note: tested using the macOS build from @semvis123 on Mojave_

**1. Start app**
success

**2. Select 'Connect to Bluetooth Device' and choose WH-1000XM4**
connection success

**3. Use the slider to change Ambient Sound Level**
ambient sound successfully changed

**4. Make another change to either Ambient Sound or the Focus on Voice option**
app crashes

The app doesn't crash for me when the only action is connecting/disconnecting the bluetooth device.

AmgC36, 2021-02-07:
> Opening this issue to gauge interest for WH-1000XM4 support. I own this model and would love to be able to use this client with it. I have zero experience in C++, but I'm happy to test and provide [...]

AmgC36, 2021-02-07:
@templeman that is a perfect description of what I am experiencing with the XM4s.

Plutoberth, 2021-02-07:
Hi,

I got reports from other people that the client works well with the XM4s using the Windows client. Could you test it there please?

mdaviesnz, 2021-02-11:
Same here with XM4.

I seem to be able to check the box for "Focus on Voice" but when unchecking it disconnects saying: "Invalid checksum!"

semvis123, 2021-02-11:
@mdaviesnz is this also on the macOS version? Are you able to test it on the Windows version?

mdaviesnz, 2021-02-11:
> @mdaviesnz is this also on the macOS version? Are you able to test it on the Windows version?

Sorry, should've mentioned that I'm testing on Windows, unable to test macOS myself.

Warchant, 2021-02-15:
I confirm that current master 3c623a8eb143854f9fd68dfa38bb93a36109a871 works with 1000WH4.

AmgC36, 2021-02-17:
> I confirm that current master 3c623a8
> works with 1000WH4.

I'm still having issues with the mac osx client with 1000WH4. When I go to
use the slider, I get the error "Unexpected error occurred and
disconnected. Invalid checksum!".

[image: image.png]
[image: image.png]

templeman, 2021-03-03:
Following up re: Windows - I was able to test the XM4s on Windows 10, and it worked well for the most part. However, the app  did fail to complete a command on more than one occasion with the `Invalid checksum!` error. This seems consistent with @mdaviesnz experience, although in my testing the error seemed to occur at random and only rarely. I left a report in #29 regarding my results.

Plutoberth, 2021-03-05:
It's baffling that the checksum is only incorrect _sometimes_, and not always || never.

Tatsh, 2021-03-17:
I am getting the checksum issue sometimes in my Qt version. I think this is mostly to do with locks/threading. The ImGui version usually works fine as it only ever allows 1 command in the air at a time.

Tatsh, 2021-03-20:
Do we have info on how the 'Speak to chat' mode is set over the 'wire' (air)?

Other than that, interested in the DSEE flag, and how to set EQ.

1AssassinSniper1, 2025-03-17:
[image]