Plutoberth/SonyHeadphonesClient #29: Headset Reports

Plutoberth, 2021-02-09:
Please post your experiences with the software here.

**Headset:**
**Client version:**
**Description:**
------

Plutoberth, 2021-02-09:
**Headset:** WH-1000-XM3
**Client version:** v1.0.2
**Description:** Fully works :)

templeman, 2021-03-03:
**Headset:** WH-1000-XM4
**Client version:** v1.1.1
**OS:** Windows 10
**Description:** Works with minor caveats.

In my testing the XM4s do work on Windows 10. I was able to successfully connect, adjust ambient sound levels, and toggle the Focus On Voice setting. I did, however, encounter an issue where the app would fail to complete a command with the `Invalid checksum!` error. This did not happen in a predictable way and I was not able to reproduce it with consistency, but it did happen more than once.

emados, 2021-03-08:
**Headset:** WH-1000-XM4
**Client version:** v1.1.1
**OS:** macOS Big Sur (11.2.2)
**Description:** I can connect but never able to adjust the ambient sound level or toggle the "Focus on Voice" option.

1. Able to successfully connect
2. I can increase the ambient sound level without getting any error, but it doesn't seem to actually change the setting on the headphones.
3. The "Focus on Voice" option is enabled, but when I check I get "Unexpected error occurred and disconnected. Invalid checksum!".
4. Reconnecting and adjusting the ambient sound level up or down raises the same error.

I didn't have the time to check the entire repo, but if there's a guide or tips on reverse engineering the Android client I may be able to help faster on this.

Plutoberth, 2021-03-11:
> **Headset:** WH-1000-XM4
> **Client version:** v1.1.1
> **OS:** macOS Big Sur (11.2.2)
> **Description:** I can connect but never able to adjust the ambient sound level or toggle the "Focus on Voice" [...]

Those issues seem very common on XM4s, but sometimes people *can* send a few messages successfully, so this is pretty perplexing.
 
> I didn't have the time to check the entire repo, but if there's a guide or tips on reverse engineering the Android client I may be able to help faster on this.

I used jadx-gui to decompile the app and export the decompilation to files. Then most of the work is to read the code and rename stuff based on your assumptions. I think that the best approach is the one described [here](https://github.com/Plutoberth/SonyHeadphonesClient/pull/36#issuecomment-795633877). I reversed it statically because I thought that I'd have to root my phone to actually sniff the messages :P

lkutsenok, 2021-03-22:
Headset: MDR-XB950BT
Client version: v1.2
Description: Fully works!

pmastrosimone, 2021-04-24:
Headset: Sony WH-1000-XM4
Client: v1.2
OS: Linux (Fedora 33)
Description: 

- Changing ambient sound seems to fully work, commands are sent successfully
- Not sure if this is implemented, but no Speak-to-Chat option

XeonForce, 2021-06-21:
**Headset:** WF-1000XM4
**Client version:** v1.2
**OS:** Windows 10
**Description:** 
Couldn't Connect: 10049

semvis123, 2021-06-21:
@XeonForce Thanks for trying, sadly the WF-1000XM4 uses a completely new protocol (literally uses ....mdr.link2 as external Accessory protocol string (still need to find the sppServiceUUID btw, but that can be found in the app)), support for these earbuds might come in the future as I already have semi working code ready (slider doesn't seem to be affecting the ambient sound), but I don't have much time to work on this rn and I don't own them, so adding support is extra difficult.

ipeacocks, 2021-06-23:
Headset: WH-1000XM3
Client version: v1.2
OS: Ubuntu 21.04
Description: Fully works, but I miss ANC turn on/off button.

Additional packages needs to be installed:
```
apt install libglfw3 libglew2.1
```

luizkowalski, 2021-06-23:
Headset: WH-CH710N
Client version: v1.2
OS: macOS Big Sur (11.4) (Chip Apple M1)

I can't connect, behavior is erratic. sometimes it closes, sometimes it says "Could not open rfcomm"

andro2157, 2021-07-21:
Headset: WH-XB900N
Client version: v1.2
OS: Windows
Description: Works perfectly!

Netanel-Lalazar, 2021-07-22:
Headset: WF-1000XM4
Client version: v1.2 probably 
OS: Windows 10
Description:
Works great! thank you! :-)

emiel556, 2021-07-27:
**Headset:** WF-1000XM4
**Client version:** v1.2
**OS:** Windows 10
**Description:** Couldn't Connect: 10049 

Headset has latest firmware (1.2.6) installed today

pumpkinlink, 2021-08-10:
**Headset:** WH-1000XM3
**Client version:** v1.2
**OS:** Ubuntu 20.04
**Description:** Fully works.

Only one additional package needs to be installed:

```sh
apt install libglfw3
```

comio, 2021-09-02:
Headset: WH-N910H
Client version: v1.2
OS: Ubuntu 21.04
Description: Fully works :)

runtav-gyz, 2021-09-04:
Headset: MDR-XB950BT
Client version: v1.2
OS: Fedora 34
Description: Error - Could not connect to bluetooth socket.

[image]

dogtopus, 2021-09-23:
**Headset**: WH-1000XM2
**Client version**: commit e671114cc9e89355d42a215241e41f25a01d3d32
**OS**: Arch Linux
**Description**: Had to downgrade C++ standard from GNU++20 to C++17 for it to compile (`SonyHeadphonesClient/Client/SingleInstanceFuture.h:15:33: error: expected unqualified-id before ‘)’ token`), but otherwise seems to work fine without any other modifications.

[image]

Casul51, 2021-11-26:
Headset: WI-1000XM2
Client version: v1.2
OS: Arch Linux
Description:
Works, but all changes to ambient sound mode need to re-enable a function few times to apply.
Had to install dependencies before use:
`sudo pamac install glfw-x11 glew-2.1`

stephaniecarag, 2021-11-28:
Headset: WF-1000XM4
Client version: v1.2
OS: Windows 10
Description: Getting an error "Couldn't connect: 10049"

[image]

dogtopus, 2021-12-17:
**Headset**: WH-1000XM4
**Client version**: commit e671114
**OS**: Arch Linux
**Description**: Finds the serial port but stuck on "sending commands..." forever. Did $\*\*y change the protocol?

CodeBrauer, 2021-12-17:
Headset: WH-1000-XM3
Client version: v1.2
OS: macOS 10.15.7
Description: Fully works on second start of the App. First start the controls did not operate the headphones.

ralex, 2022-02-01:
**Headset**: WH-1000XM2
**Client version**: v1.2
**OS**: Debian GNU/Linux 12 (Bookworm)
**Description**: Works like a charm!

[image]

Additional packages installed:
```bash
apt install libglew2.1 libglfw3 libopengl0
```

technos12, 2022-03-28:
Headset: WF-1000XM4 v1.2.6
Client version: Self-build from [e671114] and from release [v1.2]
OS: Windows10 21H1
Description: Application start, detects device, but crash at connect with answer Couldn't Connect: 10049

I would love to help, I have knowledge in programming in C#/WPF and I would really like to be able to change some parameters from a desktop App.
@semvis123 said that the used protocol is mdr.link2 but I can't find any information on that. I can make tests with my device and VSStudio.

semvis123, 2022-03-28:
> @semvis123 said that the used protocol is mdr.link2 but I can't find any information on that. I can make tests with my device and VSStudio.

`jp.co.sony.songpal.mdr.link2` seems to be the new protocol which the WF-1000XM4 uses.
Since I don't own any device that uses this new protocol, I can't really debug anything or collect any logs.
But I've managed to make it work in a Jailbreak tweak I made (based on some logs and feedback I received from users).
[See this file (apologies for the messy code)](https://github.com/semvis123/Sonitus/blob/736ccc5573fe2687d15765dbe8b72e10d068c270/Sony.mm#L25)

This implementation only sends data, so it doesn't know if it was successful or not. (for simplicity)
Because I can't test it, I can only report that this implementation was able to change the noise cancelling to ambient sound and back according to users ([with the default settings](https://github.com/semvis123/Sonitus/blob/736ccc5573fe2687d15765dbe8b72e10d068c270/Tweak.xm#L51)), iirc custom values for `ASMValue` didn't seem to work (it had to be 20).
Because this implementation is purely based on logs I am not sure if the variable names are correct for the V2 protocol.

I would suggest you to collect your own logs and see if you can find any similarities between them.
Another option would be to decompile the android app and research that, but it's quite a large app and pretty obfuscated.

technos12, 2022-03-29:
> I would suggest you to collect your own logs and see if you can find any similarities between them. Another option would be to decompile the android app and research that, but it's quite a large app and pretty obfuscated.

Cool, thank you for the answer !
I have created a working environment on my usual tools, so I can discover the device, establish a connexion, get the stream and write to it.
In parallel, I used the code from Sonitus to create the frame of data to be sent. As the code is covering both version of the protocol and all the different scenario with ? : it is sometimes difficult to follow the logic. So my headset is actually not reacting to the command.
I would like to capture the frames (collect my logs) sent from my phone (android) and use them as an example. Could you help me about this first step ?
Thanks.

audunmg, 2022-04-21:
**Headset**: WH-1000XM2
**Client version**: v1.2
**OS**: Arch Linux
**Description**: Works great!

Thank you!

rubikscraft, 2022-05-28:
Headset: WH-1000XM5
Client version: v1.3.0
OS: Fedora 36
Description: Can't connect

I get `Error: could not connect to bluetooth socket` in the gui
And
```
connecting to 88:C9:E8:16:F5:FD
channel: 0
closed
```
In the cli

So I suppose the XM5 uses a different protocol again.

corgana, 2022-07-22:
Headset: WF-1000XM4
Client version: v1.3.1
OS: Windows 10
Description: Error "Couldn't connect: 10049"

zxhzxhz, 2022-07-23:
Headset: WH-1000XM2
Client version: v1.3.1(headphone firmware version:1.0.5)
OS: Windows 10
Description: Sound Position and Surround(VPT) not working

Alessandar, 2022-07-25:
Headset: WH-CH700N
Client version: v1.3.1
OS: Arch Linux
Description: Works!

Kingfire4545, 2022-07-27:
Headset:Playstation Wireless Gold Headset
Client version:v1.31
OS:Windows 10
Description:Not detected

zzamizz, 2022-08-05:
Headset: WF-1000XM4
Client version: v1.3.1
OS: Windows 10
Description: Error "Couldn't connect: 10049"

lololo117, 2022-08-11:
Headset: WF-1000XM4
Client version: v1.3.1
OS: Windows 10
Description: Error "Couldn't connect: 10049"

forzap, 2022-08-17:
Headset: Linkbuds S
Client version: v1.3.1
OS: macOS Monterey on M1 Macbook Pro
Description: Connect to bluetooth device failed with error "Not connected, connection timed out."

IceCodeNew, 2022-08-22:
Headset: Linkbuds S
Client version: v1.3.1
OS: macOS BigSur (11.6.5) on MacBook Pro (13-inch, 2017, Two Thunderbolt 3 ports)
Description: Connect to bluetooth device failed with error "Not connected, connection timed out."

kyngs, 2022-08-31:
Headset: WH-1000XM2
Client version: 1.3.1
OS: Arch Linux
Description: Works fully

ezracelli, 2022-09-23:
Headset: WF-1000XM4
Client version: v1.3.1
OS: macOS 13.0 Ventura Beta on M1 MacBook Pro 16"
Description: Connect to bluetooth device failed with error "Not connected, connection timed out."

AKomar-sm, 2022-11-03:
Headset: WH-CH510
Client version: v1.3.1
OS: Windows 11
Description: Couldn't connect: 10049

Cancoman, 2022-11-03:
Headset: WF-1000XM4
Client version: v1.3.1
OS: Windows 11 (21H2)
Description: It doesn't want to connect (Error Code: 10049) 

My Knowledge: The Earphones are fully updated with the newest Version 1.5.0, and it makes me so sad that I'll get a depression, I never knew that there is someone skilled like you and makes an Open-Source App for Windows, and I was happy to see that and to try how it will sound after set some setting up, unfortunately it wouldn't connect with any options… Like open the app as administrator or start run it with Older OS but nothing work, I hope it will be soon fixed. I wish it for my life, I was waiting for a solution since day 1, and it is my dream that thing will work for the WF4, please. Plutoberth, I will donate you and motivate you up, that you would make me and other Users that they have the same issue as me, our dreams true. 

[image]

Cancoman, 2022-11-13:
Please, Sir, fix this issue for the WF-1000XM4, I want to access to all these setting you mentioned.

Cancoman, 2022-11-13:
Sign us with your help

MindPhaser34, 2022-12-20:
Headset: WH-1000XM5
Client version: v1.3.1
OS: macOS Ventura 13.1 on M1 Macbook Pro
Description: Connect to bluetooth device failed with error "Not connected, connection timed out."

HirIMogul, 2023-02-14:
Headset: WH-1000XM5
Client version: v1.3.1
OS: Windows 10 22H2
Description: "Couldn't connect: 10049"
[image]

jaikomodo, 2023-04-03:
Headset: Linkbuds S
Client Version: 1.3.1
OS: MacOS Ventura 13.2.1
Description: Error "Not connected, connection timed out"

Ortieez, 2023-05-18:
Headset: WH-1000-XM3
Client version: v1.3.1
OS: Fedora 38
Description: Everything works

If anyone is on fedora and is getting `./SonyHeadphonesClient: error while loading shared libraries: libGLEW.so.2.2: cannot open shared object file: No such file or directory`

Install using these: `sudo dnf install bluez-libs-devel glew-devel glfw-devel dbus-devel`

lostbow, 2023-08-03:
Headset: WF-1000XM4
Client version: v1.3.1
OS: Linux Mint
Description: Could not connect to Bluetooth spd server and Could not connect to Bluetooth socket

Started by running the below command with no luck.
sudo apt install libbluetooth-dev libglew-dev libglfw3-dev libdbus-1-dev

mrostamii, 2023-08-22:
**Headset:** WF-LS900N/BC (LinkBuds S)
**Client version:** v1.3.2
**OS:** Linux (Ubuntu 22.04.3 LTS)
**Description:** Cannot connect to the device with `Not connected, connection timed out.` error.

dantfusie, 2023-09-04:
Headset: WH-CH720N
Client version: v1.3.1
OS: Windows 11 22H2
Description: "Couldn't connect: 10049" - This is a fairly recent device so I could understand why it dosen't connect though.

QuentinAyers1917, 2023-09-26:
Headset: Linkbuds S
Client version: v1.3.2
OS: Arch Linux
Description: Error: could not connect to bluetooth socket in the gui

Also

```
connecting to F8:4E:17:73:D1:3E
channel: 0
closed
```

In the terminal

LOTGx, 2023-10-07:
Headset: WH-CH720N
Client: v1.3.1
OS: Windows 10
Description: Couldn't connect: 10049

baragatti, 2023-10-16:
Headset: WH-XB910N
Client version: v1.3.1
OS: Windows 11
Description: Couldn't Connect: 10049

egormanga, 2023-10-27:
Headset: WH-XB910N
Client version: v1.3.1
OS: Arch Linux
Description: `Error: could not connect to bluetooth socket`
```
connecting to <MAC>
channel: 0
closed
```

vojtanethio, 2023-11-07:
Headset: WF-1000XM3
Client version: v1.2
OS: Windows 10
Description:  I´m able to control Ambient Sound😊

dkuras1, 2023-11-27:
Headset: WH-CH520
Client version: v1.3.2
OS: macOS Sonoma 14.0, Macbook Air M1
Description: Connect to bluetooth device failed with error "Not connected, connection timed out."

justinlin099, 2023-11-30:
Headset: Linkbuds S
Client version: v1.3.1
OS: Windows 11
Description: Couldn't Connect: 10049

antran22, 2023-12-06:
Headset: WH-XB900N
Client version: v1.3.2
OS: MacOS Sonoma 14.0, Macbook Air M1
Description: Working out of the box for all features.

PrismaticSpirit, 2023-12-10:
Headset: WH-1000XM5
Client version: v1.3.1
OS: Windows 11
Description: Couldn't Connect 10049

SViN24, 2023-12-17:
Headset: WF-C500
Client version: 1.3.2
OS: Linux (archlinux/aur)
Description: could not connect to bluetooth socket.
Also tried disconnecting them from my PC to connect them via the app.

[image]

lmaoleonix, 2024-01-04:
Headset: WH-XB910N
Client version: v1.3.2
OS: Windows 11
Description: Couldn't Connect 10049

trippxn, 2024-01-24:
Headset: WH-CH720N
Client version: v1.3.2
OS: MacOS Sonoma Version 14.1.2
Description: "Not connected, connection timed out"

[image]

[15 more replies]