Plutoberth/SonyHeadphonesClient #33: Switch between Stereo / Hands-Free mode

Warchant, 2021-02-15:
Sony 1000WH4 can work in 2 modes: 
- **hands-free** - poor sound quality, mic works
- **stereo** - good sound quality, mic doesn't work

I would like to be able to easily switch between these, as I both listen to music and sometimes use HF for online meetings. 

Is there a switch implemented in this client?

If no, I would like to contribute and implement it.

Warchant, 2021-02-15:
Just built it. Apparently, there's no switch like that.

Plutoberth, 2021-02-15:
Correct. Two things:

1. There are prebuilt versions in the Releases page.
2. The app is built in order to support functions that are *specific* to Sony headphones. Switching between hands-free is not done using a proprietary Sony protocol, but a different device selection. I'm saying this because Zoom easily allows me to switch between the two options even though the Zoom people obviously never reverse engineered the app. So while support *can* be added, this tool is not the right place.

Warchant, 2021-02-15:
@Plutoberth for some reason I can't switch :( When I select handsfree output, it is suddenly gone. Same for input - mic simply does not work.

Everything works with mobile.

Plutoberth, 2021-02-15:
> 
> 
> @Plutoberth for some reason I can't switch :( When I select handsfree output, sound is suddenly gone. Same for input - mic simply does not work.
> 
> Everything works with mobile.

Odd. Sounds like an OS issue.

closed 2021-02-15 by Warchant

mirh, 2021-07-09:
Switching audio output endpoint on-the-fly may not work with some bugged applications
Also, selecting the microphone automatically switches the whole connection over to HFP, so if you still have selected the normal A2DP output it goes bust (even though programs aren't aware of that)