Plutoberth/SonyHeadphonesClient #35: Custom Voicepacks

KastB, 2021-02-21:
Hi,

first of all: great work, I love this project.

Secondly: Would it be possible to support custom voice packs?
You can change the language in the app and the new sound files are then uploaded to the headphones.
Could we mimic this process and provide our own files?

The primary reason for me would be a shorter startup time (Why bother me with a "power - on", I would just use a very short empty file here cause I'm always late and every second counts ;) ) and shorter breaks in a meeting (I know I can completely turn off the output, but for know I don't know the battery state then).

Did you record the data from the bluetooth connection to reverse engineer this protocol?
Can you give hints on how to start?
Do you think there is the risk of bricking the device?

Plutoberth, 2021-02-21:
Hi, I'm really glad that you like it!

Are you sure that new sound files are UPLOADED to the headphones when a language is selected, and that they're not already stored on them and then merely selected by the command?  If it's the latter, then you'll have to bypass their signature checks (if they exist) and install your own firmware somehow. It'll make for a very interesting research project, but it'll obviously take much more time :)

> Did you record the data from the bluetooth connection to reverse engineer this protocol?

At first I tried to record the data and understand it that way, but I faced some difficulties with extracting the RFCOMM data from the L2CAP sniffs that Android provides (probably would've been easier to use Frida), so I just analyzed everything statically. I think that it could be done with a bit more effort, and then we could answer the question posed in the beginning of this comment.

> Can you give hints on how to start?

I just took the app, decompiled all of it using jadx, deleted some external packages that weren't related, and then got to work. They used ProGuard, so most of the names were stripped, but other stuff like enum names and (of course) system lib calls are still fully visible. Unfortunately I can't publish or otherwise share the annotated code with you - I don't want to get in trouble. 

>  Do you think there is the risk of bricking the device?

I think that the chance of bricking the device (unless the firmware is replaced) is very low, but it's always a possibility.

Please update me :D

semvis123, 2021-02-21:
This is something I personally also wanted, because I don't like the low battery warnings that stop the audio for 3 seconds :)

I suppose the firmware is updated or the language pack is sent or something like that, because it takes a long time before it is done, so it probably isn't some variable that you can change to select a different language. And since I can't seem to find any audio files that contain those languages in the apk, it seems like it is downloaded from a server.

KastB, 2021-02-21:
Because of the delay and the size estimate for all languages I guessed it's not just a variable either.

@semvis123 Great to hear that someone else is also interested (and annoyed) in this topic.
Your finding, that there are no such files in the apk is pretty cool. Maybe we can use that with a fake server to use the original app similar to the downgrade process of the firmware (https://www.reddit.com/r/sony/comments/akynjz/psa_the_new_sony_wh1000xm3_firmware_v411_is/ekcytlg/?utm_source=share&utm_medium=web2x).
I hope we can still use this issue to discuss ideas even if we come up with a solution outside of code in here...

I will probably not go the way of decompiling it - I tried it for other projects and failed most of the time with that. If I find time I would try to sniff the bt and wifi connection first and hope to get an idea of how it works there first...

Cheers

semvis123, 2021-02-21:
Just got the url for the bin file that will be uploaded to the headset (don't think that I'm able to legally share the url lol), I used Mitmproxy to get it. It seems like the bin file is just a mp3, you can listen to it and you will hear the spoken words, the file is language specific, so it contains only that language you chose.

There are empty noticeable start and end markers for each sound (opened the file with a hex editor), including mp3 metadata?, so I suppose that you should be able to change the sound lengths.

I'm not sure how to modify such a file, but in theory it should be possible I guess.

Plutoberth, 2021-02-21:
@semvis123 Awesome, so sounds like changing the sounds should be very easy. At this point making it a separate tool that mitms the file request would probably be WAY easier than figuring out the bluetooth commands and the actual encoding.

Btw random question @KastB , do you know if the newest firmware for the XM3s still has the ANC issues?

KastB, 2021-02-22:
@Plutoberth unfortunatelly I can't really tell. I got my headphones very recently and never experienced the 2.0 firmware so I have no comparison. Downgrade does seem possible only for 4.1 and I already have 4.5.2.

semvis123, 2021-02-22:
hmm seems like the bin file is a otau file, but I don't think that creating those files is made public.
https://developer.qualcomm.com/qfile/34081/csr102x_otau_overview.pdf
Looks pretty difficult to create such a file manually (at least for me😅).

KastB, 2021-02-22:
From those document it seems that no encryption/signature is natively included at least

KastB, 2021-02-22:
@semvis123 I tried mitmproxy but all I can see are connectivitychecks of google.
Mobile data is deactivated.
I deleted cache and data of the app to force re-downloading of any language file.
Can you give me a hint?

semvis123, 2021-02-22:
> @semvis123 I tried mitmproxy but all I can see are connectivitychecks of google.
> 
> Mobile data is deactivated.
> 
> I deleted cache and data of the app to force re-downloading of any language file.
> 
> Can you give me a hint?

I had to install a root certificate, by going to `mitm.it` and follow the instructions there. If you are scared of the mitmproxy cli you can use `mitmweb` which should come with mitmproxy.

You should be able to see many requests after installing the certificate correctly, for example on each menu that you open in the sonyheadphones app it sends a log to Sony😬

KastB, 2021-02-24:
For me, there was certificate pinning active in the sony app. apk-mitm can mitigate this, but otherwise there is no traffic visible for mitmproxy...

Plutoberth, 2021-09-19:
Just analyzed it and it looks like it's just a bunch of MP3s. Could be pretty easy, unless you want to make the voice clips *longer*. I still want to figure out how the file is sent to the headphones - requiring each user to run apk-mitm, mitm the transfer, and replace the file is a bit of a hassle. Maybe the OTAU thing is a good start.

Casul51, 2021-11-25:
Any progress on that?

Plutoberth, 2021-11-26:
> Any progress on that?

No, maybe when some time frees up for me in a few months.

lapaz17, 2022-01-25:
Guys, hoe can i see these otau(mp3) files inside earphone after connecting it to pc?

KastB, 2022-01-25:
Noone I guess. It should be in the network stream when you would wiretap the app.

ppamorim, 2022-02-09:
@Plutoberth I would cross the atlantic to remove the annoying low battery warning. I will f do it. I will grap the decompiler and fix that ****.

semvis123, 2022-02-13:
Managed to mute a beep in one of the mp3's (power off sound).
I used [mp3DirectCut](https://mpesch3.de) to alter a mp3 chunk (extracted with [Veles](https://github.com/codilime/veles)) of the binary file without changing it's length, then I pasted this muted mp3 chunk back into the binary file.
To upload it to the headphones I used [Mitmproxy](https://mitmproxy.org) (mitmweb) to intercept and replace the info.xml, disclaimer.xml and the firmware.bin.
The encrypted xml files were created using a few functions of [this helpful project (xm3utilitytool)](https://github.com/elmarto/xm3utilitytool)

semvis123, 2022-02-14:
Also managed to shorten the mp3's (I tried the bluetooth connected and power_on sounds this time), I just replaced the remaining bytes with zeros to keep the mp3 chunk at the same length.

sashkyn, 2023-09-28:
@semvis123 do you need help to implement it inside MacOS app?
I'm iOS developer but I don't know ObjC properly 😄 

I'm really interested in resolving this issue!

semvis123, 2023-09-28:
@sashkyn I currently don't have time to work on this project. 

Implementing this feature would also be quite a challenge. It requires the updating protocol to be reverse engineered. 
A simpler implementation would be to request a binary file, parse the mp3s, request new mp3s, and modify the binary, and output it. 
But that would still require the user to intercept the file and replace it during an update in the app. Automating the proxy might also be possible, but will also complicate things.

vrubleg, 2024-03-08:
I'd also like to replace/remove sounds in the WH-1000XM4 firmware, but seems like its update files are compressed or encrypted now.

https://info.update.sony.net/HP002/MDRID294300/contents/0007/UPG_2.5.0_20210413_enc.bin

Seems like all newer Sony headphones use this new format. Did anybody try to reverse engineer and decompress/decrypt it?

mirh, 2024-09-30:
These guys also tried tinkering
https://github.com/iclemens/bose
https://github.com/egaebel/sony-headphones-hack/issues/1
Btw while the last one may appreciate some hardware hacking assistance, let's just say that qualcomm's Audio Development Kit isn't as private and difficult to find as it once was.