Plutoberth/SonyHeadphonesClient #44: Hide macOS dock icon

lkutsenok, 2021-03-22:
Hi!
I've managed to hide the dock icon, while keeping the menu bar. For me, the dock icon doesn't make much sense, cause the  best thing about the macOS app is the quick toggle between noise cancelling and ambient mode. 
I have little experience with native Objective-C, but managed to hide the dock icon using following modifications:

- in info.plist added `LSUIElement String 1`
- in AppDelegate.mm added
```
- (BOOL)applicationShouldHandleReopen:(NSApplication *)theApplication hasVisibleWindows:(BOOL)flag
{
    //new
    [NSApp setActivationPolicy:NSApplicationActivationPolicyRegular];
    [NSApp setPresentationOptions:NSApplicationPresentationDefault];
    [NSMenu setMenuBarVisible:NO];
    [NSMenu setMenuBarVisible:YES];
    // end
    if (flag) {
        return NO;
    }
    else {
        [_window makeKeyAndOrderFront:self];
        return YES;
    }
}
```

Maybe we can manage the settings via menu bar?

Plutoberth, 2021-03-22:
@semvis123 is the expert here

vincent-le-normand, 2021-03-23:
Hi,

I created a similar software for the Parrot Zik headphone, and now I own a Sony headset. If you need to, I can help on this project:
https://github.com/vincent-le-normand/Parrot-Status

I tends to believe that it would be better to only offer one option: either show the dock icon for everybody, or for nobody. I'd vote for no dock icon, and the UI embedded in the menu, something similar to Big Sur control center.

If you want to change this setting programatically, it appears you should perform this way:
https://stackoverflow.com/questions/5382932/how-to-create-a-helper-application-lsuielement-that-also-has-a-removable-doc

semvis123, 2021-03-23:
Hi, 
hiding the dock icon might be preferable for most users, but maybe other people want that fine control of the current gui interface. We could create some user preferences that could store things like, 
* dock icon visible,
* menu bar icon visible,
* menu bar icon quick toggle (as currently default)
* menu bar full gui version,  all settings move to the menu bar, as @vincent-le-normand suggested, but this has a disadvantage because it doesn't provide the speed of the quick toggle. 
* start-at-login
* auto (re)connect to known Mac address (could configured with the native bluetooth ui)
* NC quick toggle value, this value is used for the Noise Cancelation quick toggle
* ASM quick toggle value, this value is used for the Ambient sound (transparency) quick toggle

https://developer.apple.com/documentation/foundation/nsuserdefaults?language=objc

Those are just some ideas, and I currently do not have enough time to fulfill this all. But I would love to hear your opinion about something like this.

vincent-le-normand, 2021-03-23:
This could be an option. However, some settings are not compatible (it's not a good idea to disable both dock icon and menu icon).
Regarding the menu bar full gui version, maybe it's possible to have the quick toggle and just nearby another icon that offer the full setting view. Some Apple Menu Items (such as date time) also offer quick toggles if you press options while clicking, but it's not really discoverable… Also, we could offer a quick toggle using a global hot key. 

Also, I hope someday we would have a way to display battery level in the menu. 

The other options (start at login, auto reconnect, …) would be really great! 

I am quite busy right now, be maybe I can help someday…

semvis123, 2021-03-23:
> This could be an option. However, some settings are not compatible (it's not a good idea to disable both dock icon and menu icon).

Yeah I know, but we should just disable the other button, so that you won't end up with a broken program or we could use a NSSwitch, but that is only available in the more recent macOS versions.

Don't know about both icons, there is already enough clutter in my menu bar :)
Long pressing the menu button might also be an option,  but I don't know if this is possible, and it isn't intuitive for the user.

Global hot key would indeed be great, battery level could be added when receiving the settings/information from the headphones is implemented.

lkutsenok, 2021-03-23:
> Global hot key would indeed be great

I have some ideas on how to implement it with Automator, I'll try them later and post the results

CodeBrauer, 2021-12-17:
If there would be an CLI option for any control, it would be probably the best.

Then you can use it however you want:

- Alfred / LaunchBar
- BetterTouchTools
- https://xbarapp.com
- https://github.com/sveinbjornt/Platypus
- and many more...