Plutoberth/SonyHeadphonesClient #51: Compiling issues (Arch Linux, Gnome, Wayland)

dries007, 2021-07-23:
Hi, I'm having trouble compiling this package.

I cannot use the precompiled release, as Arch is on GLEW 2.2, and I'm using Wayland (which requires GLEW to be compiled with some options), so simply downloading an old version is not that straight forward.

So I'm trying now to compile from source and I'm getting the following log.

```
[dries@DriesPC build]$ cmake  ..
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3") 
CMake Warning (dev) at /usr/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (DBUS) does
  not match the name of the calling package (DBus).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  linux/FindDBus.cmake:59 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:58 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found DBUS: /usr/include/dbus-1.0;/usr/lib/dbus-1.0/include  
-- Found GLEW: /usr/include (found version "2.2.0") 
-- Found OpenGL: /usr/lib/libOpenGL.so   
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dries/Projects/SonyHeadphonesClient/Client/build
[dries@DriesPC build]$ cmake --build .
[  5%] Building CXX object CMakeFiles/SonyHeadphonesClient.dir/BluetoothWrapper.cpp.o
[ 11%] Building CXX object CMakeFiles/SonyHeadphonesClient.dir/ByteMagic.cpp.o
[ 17%] Building CXX object CMakeFiles/SonyHeadphonesClient.dir/CommandSerializer.cpp.o
[ 23%] Building CXX object CMakeFiles/SonyHeadphonesClient.dir/TimedMessageQueue.cpp.o
[ 29%] Building CXX object CMakeFiles/SonyHeadphonesClient.dir/CrossPlatformGUI.cpp.o
In file included from /home/dries/Projects/SonyHeadphonesClient/Client/CrossPlatformGUI.h:10,
                 from /home/dries/Projects/SonyHeadphonesClient/Client/CrossPlatformGUI.cpp:1:
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h:15:33: error: expected unqualified-id before ‘)’ token
   15 |         SingleInstanceFuture<T>() = default;
      |                                 ^
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h:21:48: error: invalid declarator before ‘other’
   21 |         SingleInstanceFuture<T>(std::future<T> other);
      |                                                ^~~~~
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h:21:47: error: expected ‘)’ before ‘other’
   21 |         SingleInstanceFuture<T>(std::future<T> other);
      |                                ~              ^~~~~~
      |                                               )
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h:48:8: error: no declaration matches ‘SingleInstanceFuture<T>::SingleInstanceFuture(std::future<_Res>)’
   48 | inline SingleInstanceFuture<T>::SingleInstanceFuture(std::future<T> other) : std::future<T>(std::move(other)) {}
      |        ^~~~~~~~~~~~~~~~~~~~~~~
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h:48:8: note: no functions named ‘SingleInstanceFuture<T>::SingleInstanceFuture(std::future<_Res>)’
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h:12:7: note: ‘class SingleInstanceFuture<T>’ defined here
   12 | class SingleInstanceFuture : public std::future<T>
[... 13 lines cut ...]
      |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h:51:83: note:   initializing argument 1 of ‘SingleInstanceFuture<T> SingleInstanceFuture<T>::operator=(std::future<_Res>&) [with T = std::vector<BluetoothDevice>]’
   51 | inline SingleInstanceFuture<T> SingleInstanceFuture<T>::operator=(std::future<T>& other)
      |                                                                   ~~~~~~~~~~~~~~~~^~~~~
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h: In instantiation of ‘void SingleInstanceFuture<T>::setFromAsync(Func, Args&& ...) [with Func = CrossPlatformGUI::_drawASMControls()::<lambda()>; Args = {}; T = int]’:
/home/dries/Projects/SonyHeadphonesClient/Client/CrossPlatformGUI.cpp:223:41:   required from here
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h:33:15: error: cannot bind non-const lvalue reference of type ‘std::future<int>&’ to an rvalue of type ‘std::future<int>’
   33 |         *this = std::async(std::launch::async, func, std::forward<Args>(args)...);
      |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h:51:83: note:   initializing argument 1 of ‘SingleInstanceFuture<T> SingleInstanceFuture<T>::operator=(std::future<_Res>&) [with T = int]’
   51 | inline SingleInstanceFuture<T> SingleInstanceFuture<T>::operator=(std::future<T>& other)
      |                                                                   ~~~~~~~~~~~~~~~~^~~~~
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h: In instantiation of ‘void SingleInstanceFuture<T>::setFromAsync(Func, Args&& ...) [with Func = CrossPlatformGUI::CrossPlatformGUI(BluetoothWrapper)::<lambda()>; Args = {}; T = std::vector<BluetoothDevice>]’:
/home/dries/Projects/SonyHeadphonesClient/Client/CrossPlatformGUI.cpp:245:44:   required from here
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h:33:15: error: cannot bind non-const lvalue reference of type ‘std::future<std::vector<BluetoothDevice> >&’ to an rvalue of type ‘std::future<std::vector<BluetoothDevice> >’
   33 |         *this = std::async(std::launch::async, func, std::forward<Args>(args)...);
      |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/dries/Projects/SonyHeadphonesClient/Client/SingleInstanceFuture.h:51:83: note:   initializing argument 1 of ‘SingleInstanceFuture<T> SingleInstanceFuture<T>::operator=(std::future<_Res>&) [with T = std::vector<BluetoothDevice>]’
   51 | inline SingleInstanceFuture<T> SingleInstanceFuture<T>::operator=(std::future<T>& other)
      |                                                                   ~~~~~~~~~~~~~~~~^~~~~
make[2]: *** [CMakeFiles/SonyHeadphonesClient.dir/build.make:132: CMakeFiles/SonyHeadphonesClient.dir/CrossPlatformGUI.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/SonyHeadphonesClient.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
[dries@DriesPC build]$ 
```

dries007, 2021-07-23:
I just tried with clang as a last ditch effort that then it does compile, so it seems to be an issue with the use of GCC 11.1.

This is what the app looks like, but I suppose that's more an issue with the img gui library than your project:
[image]

And it behaves a bit strange when re-sizing. But still usable :)

Plutoberth, 2021-09-17:
I'm not sure what the issue is. It kinda looks like the compiler doesn't agree with one of the features used, but GCC 11 is really new so it's unlikely. Glad that it worked for you with clang :)
I'm keeping this open in-case anyone else has the same issue.

dogtopus, 2021-09-23:
Crosspost from the note I left in #29: Downgrading C++ standard to C++17 seems to fix the issue. Could be that GNU++20 is having some problem with `std::future`?

fridlmue, 2021-11-29:
@Tatsh May be a kind of "noob question", but could you elaborate a little if I have to set `=std=c++20` somehow in the CMAKE config or execute it with `cmake --build .` or how to correctly apply it?

dogtopus, 2021-11-29:
@fridlmue It should be the default to C++20.

Zerogoki00, 2021-12-03:
Same issue

Tatsh, 2021-12-03:
@fridlmue `Client/SingleInstanceFuture.h` lines 15 and 22:

```patch
diff --git a/Client/SingleInstanceFuture.h b/Client/SingleInstanceFuture.h
index 8af733f..d2e6c49 100644
--- a/Client/SingleInstanceFuture.h
+++ b/Client/SingleInstanceFuture.h
@@ -12,13 +12,13 @@ template <class T>
 class SingleInstanceFuture : public std::future<T>
 {
 public:
-	SingleInstanceFuture<T>() = default;
+	SingleInstanceFuture(void) = default;
 	template<class Func, class... Args>
 	void setFromAsync(Func func, Args&&... args) noexcept(false);
 	bool ready();
 
 private:
-	SingleInstanceFuture<T>(std::future<T> other);
+	SingleInstanceFuture(std::future<T> other);
 	SingleInstanceFuture<T> operator=(std::future<T>& other);
 };
```

GCC 11 will accept this now with `-std=gnu++20` (the default of the project). Clang will also accept this syntax (tested with version 13).

Similar issue in another project: https://github.com/boostorg/msm/pull/26/

Related mailing list post: https://gcc.gnu.org/pipermail/gcc-patches/2020-April/543301.html

Zerogoki00, 2022-02-09:
@Tatsh thank you, your patch works

PR #75 (2022-06-03): fix gcc compilation
Integrate the patch by @Tatsh. Closes #51.
closed 2022-06-03 by Plutoberth