Plutoberth/SonyHeadphonesClient #53: Arch - could not connect to bluetooth socket

Mhsbrian, 2021-07-27:
Running Arch here, got past the glew 2.1 errors originally and now the application runs fine as expected but when I select WF-1000XM4 I receive "Could not connect to bluetooth socket"

Bluez-git 5.60
bluetooth-support 1-3
bluez-qt 5.84

[image]

Tried a few things but can't seem to get past this point but no luck, suggestions?

semvis123, 2021-07-27:
The WF-1000XM4's are currently unsupported, because it uses a different protocol.

Mhsbrian, 2021-07-27:
> The WF-1000XM4's are currently unsupported, because it uses a different protocol.

Ah,

Understood, thank you much for the quick reply. 

Looking forward to seeing this improve, good work!

closed 2021-07-27 by Mhsbrian

maxlath, 2022-03-02:
for the record, same error with the MDR-1000X model