Plutoberth/SonyHeadphonesClient #55: WF-1000XM4 support

mightysashiman, 2021-08-17:
Hi, tried connecting to my WF-1000XM4's from the Mac app. App crashes. I also have an android phone with Sony's headphone app.
Please let me know if there is some logs / script I can provide / run to help reverse engineer the device communication protocol.
Kind regards

Plutoberth, 2022-05-21:
Hi,

I'm not so sure about the Mac app, as I didn't develop it myself (I don't have a Mac). Could you test using the windows app?

Plutoberth, 2022-05-21:
@semvis123  might be relevant here

semvis123, 2022-05-21:
It isn't necessarily MacOS specific, the **WF**-1000XM4's are just not implemented yet.

You could collect logs using your Android phone (don't exactly know if root is required).

To do that you need to enable the `Bluetooth HCI Snoop log` in the developer options (which you can enable by pressing the build version number in settings a few times iirc).  Once enabled, you need to toggle Bluetooth to apply it.

Once you have done that, you can start collecting the logs (open the app and start toggling switches and sliding sliders, preferably in a known order).

When you're done collecting, you can retrieve the log file with ADB (details seem to vary by device).
USB debugging should probably be enabled.

The file name should be: `btsnoop_hci.log`

This file can be opened with Wireshark for inspection.

I managed to perform these steps on a rooted Android 10 device.

Sources:
https://www.xda-developers.com/install-adb-windows-macos-linux/
https://www.nowsecure.com/blog/2017/02/07/bluetooth-packet-capture-on-android-4-4/
https://stackoverflow.com/questions/28445552/bluetooth-hci-snoop-log-not-generated#30352487

1oh1, 2022-07-17:
I added some logs (only toggled Speak to Chat feature) from my Android phone (Poco F1) while it was connected to the XM4s. I'll add more logs later.

Ref: https://github.com/Plutoberth/SonyHeadphonesClient/issues/65

Plutoberth, 2023-04-26:
See #65

closed 2023-04-26 by Plutoberth