Plutoberth/SonyHeadphonesClient #57: Qt Version - macOS

Plutoberth, 2021-09-17:
The Qt version currently doesn't compile on macOS. Technically this isn't Qt's fault, it's just that previously CMake wasn't used so it doesn't link IOBluetooth (and maybe other things 🤡) properly

semvis123, 2021-09-23:
After reverting this commit 25a1cd6a0aa79f6029554aad3446f3baecb694b6, it compiles just fine.
Don't know where the duplicates are that this commit was supposed to remove.
(btw, QT version works alright on macOS with the WH-1000XM3s, however sometimes it silently fails to send the command after sliding the slider)

Plutoberth, 2021-09-24:
> After reverting this commit [25a1cd6](https://github.com/Plutoberth/SonyHeadphonesClient/commit/25a1cd6a0aa79f6029554aad3446f3baecb694b6), it compiles just fine.

Ah, of course! Did just that and it works now :)

> sometimes it silently fails to send the command after sliding the slider

Right, moving to qt changed the design concept so a few things went wrong.. I opened #60 to track this.

PR #59 (2022-03-12): Fix macOS qt issues
Fixes #57
closed 2021-09-24 by Plutoberth