Plutoberth/SonyHeadphonesClient #63: Question - will EQ functionality be added soon?

denisbarac, 2022-01-06:
The only thing I've used the Sony Connect app on my phone was for the EQ. I have multiple sets of headphones because I use to do music production, so sound is quite important to me. I've set the EQ on my app, but the settings don't propagate on my laptop and PC. So....will EQ support be added? It would be greatly appreciated

Plutoberth, 2022-01-06:
I'm very busy, so it's unlikely that I'll get to it soon. If someone else wants to do it, I'll probably have the time to check and approve the PR.

V02460, 2022-02-21:
Could somebody with access to the app find out the commands for adjusting the EQ? https://github.com/Plutoberth/SonyHeadphonesClient/pull/36#issuecomment-795633877 seems to be helpful. With that info I’d give implementing EQ a try.

semvis123, 2022-02-21:
@V02460 I have logged the traffic of the iOS app (using a jailbroken device). This is using the ExternalAccessory protocol, which seems to wrap the data in a few extra bytes but the actual data should be the same.

The traffic below is switching from the vocal to the treble preset.
```

2022-02-21 21:41:02.299 MDRPlugin[76503:30918254] [EALogger] jp.co.sony.songpal.mdr.link | 13 ⬆ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x04 0x58 0x01 0x15 0x00 0x7e 0x3c 
2022-02-21 21:41:02.354 MDRPlugin[76503:30912730] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-21 21:41:02.368 MDRPlugin[76503:30912730] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬇ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x0a 0x59 0x01 0x15 0x06 0x0a 0x0a 0x0a 0x0c 0x10 0x14 0xffffffda 0x3c 
2022-02-21 21:41:02.369 MDRPlugin[76503:30919188] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-21 21:41:02.370 MDRPlugin[76503:30918254] [EALogger] jp.co.sony.songpal.mdr.link | 11 ⬆ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x02 0x5a 0x01 0x6a 0x3c 
2022-02-21 21:41:02.395 MDRPlugin[76503:30912730] [EALogger] jp.co.sony.songpal.mdr.link | 13 ⬇ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x04 0x17 0x00 0x00 0x02 0x29 0x3c 
2022-02-21 21:41:02.396 MDRPlugin[76503:30918254] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-21 21:41:02.397 MDRPlugin[76503:30912730] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-21 21:41:02.414 MDRPlugin[76503:30912730] [EALogger] jp.co.sony.songpal.mdr.link | 31 ⬇ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x15 0x5b 0x01 0x06 0x10 0x00 0x01 0x01 0x01 0xffffff90 0x01 0x03 0xffffffe8 0x01 0x09 0xffffffc4 0x01 0x18 0xffffff9c 0x01 0x3d 0x2e 0xffffff80 0x55 0x3c 
2022-02-21 21:41:02.415 MDRPlugin[76503:30919605] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 

```

V02460, 2022-02-22:
Great you got on to it that fast, thank you very much! :) EA is just for session management and has no knowledge of the transmitted data, so I think what we see is the raw stream data. Do you know what the number in front of each arrow means?

I assume this is a proprietary protocol, so more examples would be very helpful for understanding what’s going on. For starters, it would be interesting to see what changes in the messages if you use the manual preset and adjust one slider of the EQ at a time. Can you try capturing that?

semvis123, 2022-02-26:
> Do you know what the number in front of each arrow means?

It is the length of the data (the amount of bytes)

The following data captured is:
Switching off and on ambient sound control (could be used to find the difference between external accessory and 'normal' rfcomm)

Following with switching through each EQ preset: Bright, Exited, Mellow, Relaxed, ..., custom 2, off.

Manual is set with flat profile clear bass +8, custom and custom 2 are both flat and clear bass 0

There is an approximate 10 second delay between each action.

```
2022-02-26 23:27:00.787 MDRPlugin[6409:33607012] [EALogger] jp.co.sony.songpal.mdr.link | 17 ⬆ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x08 0x68 0x02 0x00 0x02 0x02 0x01 0x00 0x00 0xffffff83 0x3c 
2022-02-26 23:27:00.850 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-26 23:27:00.865 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 17 ⬇ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x08 0x69 0x02 0x00 0x02 0x02 0x01 0x00 0x00 0xffffff85 0x3c 
2022-02-26 23:27:00.866 MDRPlugin[6409:33607432] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-26 23:27:04.934 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 13 ⬇ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x04 0x13 0x00 0x14 0x00 0x37 0x3c 
2022-02-26 23:27:04.935 MDRPlugin[6409:33607012] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-26 23:27:12.392 MDRPlugin[6409:33607012] [EALogger] jp.co.sony.songpal.mdr.link | 17 ⬆ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x08 0x68 0x02 0x01 0x02 0x02 0x01 0x00 0x00 0xffffff85 0x3c 
2022-02-26 23:27:12.431 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-26 23:27:12.446 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 17 ⬇ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x08 0x69 0x02 0x01 0x02 0x02 0x01 0x00 0x00 0xffffff86 0x3c 
2022-02-26 23:27:12.447 MDRPlugin[6409:33607012] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-26 23:27:24.873 MDRPlugin[6409:33607574] [EALogger] jp.co.sony.songpal.mdr.link | 13 ⬆ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x04 0x58 0x01 0x10 0x00 0x79 0x3c 
2022-02-26 23:27:24.911 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-26 23:27:24.927 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬇ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x0a 0x59 0x01 0x10 0x06 0x09 0x0a 0x0f 0x11 0x11 0x13 0xffffffdd 0x3c 
2022-02-26 23:27:24.928 MDRPlugin[6409:33607574] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-26 23:27:24.928 MDRPlugin[6409:33607580] [EALogger] jp.co.sony.songpal.mdr.link | 11 ⬆ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x02 0x5a 0x01 0x6a 0x3c 
2022-02-26 23:27:24.990 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 13 ⬇ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x04 0x17 0x00 0x00 0x02 0x2a 0x3c 
2022-02-26 23:27:24.991 MDRPlugin[6409:33607580] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-26 23:27:24.992 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-26 23:27:25.000 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 31 ⬇ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x15 0x5b 0x01 0x06 0x10 0x00 0x01 0x01 0x01 0xffffff90 0x01 0x03 0xffffffe8 0x01 0x09 0xffffffc4 0x01 0x18 0xffffff9c 0x01 0x3d 0x2e 0xffffff80 0x54 0x3c 
2022-02-26 23:27:25.001 MDRPlugin[6409:33607574] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-26 23:27:34.742 MDRPlugin[6409:33607634] [EALogger] jp.co.sony.songpal.mdr.link | 13 ⬆ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x04 0x58 0x01 0x11 0x00 0x7a 0x3c 
2022-02-26 23:27:34.782 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-26 23:27:34.797 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬇ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x0a 0x59 0x01 0x11 0x06 0x12 0x09 0x0b 0x0a 0x0d 0x0f 0xffffffd4 0x3c 
2022-02-26 23:27:34.798 MDRPlugin[6409:33607667] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-26 23:27:34.799 MDRPlugin[6409:33607634] [EALogger] jp.co.sony.songpal.mdr.link | 11 ⬆ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x02 0x5a 0x01 0x6a 0x3c 
2022-02-26 23:27:34.907 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 13 ⬇ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x04 0x17 0x00 0x00 0x02 0x29 0x3c 
2022-02-26 23:27:34.908 MDRPlugin[6409:33607667] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-26 23:27:34.911 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-26 23:27:34.917 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 31 ⬇ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x15 0x5b 0x01 0x06 0x10 0x00 0x01 0x01 0x01 0xffffff90 0x01 0x03 0xffffffe8 0x01 0x09 0xffffffc4 0x01 0x18 0xffffff9c 0x01 0x3d 0x2e 0xffffff80 0x55 0x3c 
2022-02-26 23:27:34.918 MDRPlugin[6409:33607634] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
[... cut ...]
2022-02-26 23:29:32.149 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-26 23:29:32.163 MDRPlugin[6409:33606433] [EALogger] jp.co.sony.songpal.mdr.link | 31 ⬇ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x15 0x5b 0x01 0x06 0x10 0x00 0x01 0x01 0x01 0xffffff90 0x01 0x03 0xffffffe8 0x01 0x09 0xffffffc4 0x01 0x18 0xffffff9c 0x01 0x3d 0x2e 0xffffff80 0x55 0x3c 
2022-02-26 23:29:32.163 MDRPlugin[6409:33608086] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 

```

V02460, 2022-02-28:
Nice, we’re getting somewhere! :D I removed all ACK packets from the data and annotated the rest. The model of the equalizer settings seem to be 6 values between 0 and 20, representing the -10 to +10 range from the app. For some reason, the app lists the bass separately, but it’s uniform at the protocol level. As soon as any preset is activated, it reports an UpscalingEffectStatus of INVALID. The one packet with the irregular timestamp also informed me that you were testing your headphones while on battery, which was already relatively low ;)

For the record regarding the protocol, many packets have different payloads depending on a leading discriminating field. Lists are prefixed with a byte giving the number of elements.

As far as I can see, your trace did not capture any actual changes to the manual/custom EQ settings, though. Also, any idea what EBB stands for?

```
⬆ 0 NCASM_SET_PARAM(NC_ASM_INQUIRED_TYPE.NOISE_CANCELLING_AND_AMBIENT_SOUND_MODE, NC_ASM_EFFECT.OFF, …)
⬇ 1 NCASM_NTFY_PARAM(NC_ASM_INQUIRED_TYPE.NOISE_CANCELLING_AND_AMBIENT_SOUND_MODE, NC_ASM_EFFECT.OFF, …)
⬇ 0 COMMON_NTFY_BATTERY_LEVEL(BatteryInquiredType.BATTERY, 0x14, BatteryChargingStatus.NOT_CHARGING)
⬆ 1 NCASM_SET_PARAM(NC_ASM_INQUIRED_TYPE.NOISE_CANCELLING_AND_AMBIENT_SOUND_MODE, NC_ASM_EFFECT.ON , …)
⬇ 1 NCASM_NTFY_PARAM(NC_ASM_INQUIRED_TYPE.NOISE_CANCELLING_AND_AMBIENT_SOUND_MODE, NC_ASM_EFFECT.ON , …)

⬆ 0 EQEBB_SET_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.BRIGHT, [])
⬇ 0 EQEBB_NTFY_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.BRIGHT, [0x09, 0x0a, 0x0f, 0x11, 0x11, 0x13]
⬆ 1 EQEBB_GET_EXTENDED_INFO(EqEbbInquiredType.PRESET_EQ)
⬇ 1 COMMON_NTFY_UPSCALING_EFFECT(CommonCapabilityInquiredType.FIXED_VALUE, UpscalingEffectType.DSEE_HX, UpscalingEffectStatus.INVALID)
⬇ 0 EQEBB_RET_EXTENDED_INFO(EqEbbInquiredType.PRESET_EQ, [(EqBandInformationType.SPECIFIC_INFORMATION, 0x00 0x01), (EqBandInformationType.SPECIFIC_INFORMATION, 0x01 0x90), (EqBandInformationType.SPECIFIC_INFORMATION, 0x03 0xe8), (EqBandInformationType.SPECIFIC_INFORMATION, 0x09 0xc4), (EqBandInformationType.SPECIFIC_INFORMATION, 0x18 0x9c), (EqBandInformationType.SPECIFIC_INFORMATION, 0x3e 0x80)]

⬆ 0 EQEBB_SET_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.EXCITED, [])
⬇ 1 EQEBB_NTFY_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.EXCITED, [0x12, 0x09, 0x0b, 0x0a, 0x0d, 0x0f]
⬆ 1 EQEBB_GET_EXTENDED_INFO(EqEbbInquiredType.PRESET_EQ)
⬇ 0 COMMON_NTFY_UPSCALING_EFFECT(CommonCapabilityInquiredType.FIXED_VALUE, UpscalingEffectType.DSEE_HX, UpscalingEffectStatus.INVALID)
⬇ 1 EQEBB_RET_EXTENDED_INFO(EqEbbInquiredType.PRESET_EQ, [(EqBandInformationType.SPECIFIC_INFORMATION, 0x00 0x01), (EqBandInformationType.SPECIFIC_INFORMATION, 0x01 0x90), (EqBandInformationType.SPECIFIC_INFORMATION, 0x03 0xe8), (EqBandInformationType.SPECIFIC_INFORMATION, 0x09 0xc4), (EqBandInformationType.SPECIFIC_INFORMATION, 0x18 0x9c), (EqBandInformationType.SPECIFIC_INFORMATION, 0x3e 0x80)]

⬆ 0 EQEBB_SET_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.MELLOW, [])
⬇ 0 EQEBB_NTFY_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.MELLOW, [0x07, 0x09, 0x08, 0x07, 0x06, 0x04]
⬆ 1 EQEBB_GET_EXTENDED_INFO(EqEbbInquiredType.PRESET_EQ)
⬇ 1 COMMON_NTFY_UPSCALING_EFFECT(CommonCapabilityInquiredType.FIXED_VALUE, UpscalingEffectType.DSEE_HX, UpscalingEffectStatus.INVALID)
⬇ 0 EQEBB_RET_EXTENDED_INFO(EqEbbInquiredType.PRESET_EQ, [(EqBandInformationType.SPECIFIC_INFORMATION, 0x00 0x01), (EqBandInformationType.SPECIFIC_INFORMATION, 0x01 0x90), (EqBandInformationType.SPECIFIC_INFORMATION, 0x03 0xe8), (EqBandInformationType.SPECIFIC_INFORMATION, 0x09 0xc4), (EqBandInformationType.SPECIFIC_INFORMATION, 0x18 0x9c), (EqBandInformationType.SPECIFIC_INFORMATION, 0x3e 0x80)]

⬆ 0 EQEBB_SET_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.RELAXED, [])
⬇ 1 EQEBB_NTFY_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.RELAXED, [0x01, 0x07, 0x09, 0x07, 0x05, 0x02]
⬆ 1 EQEBB_GET_EXTENDED_INFO(EqEbbInquiredType.PRESET_EQ)
⬇ 0 COMMON_NTFY_UPSCALING_EFFECT(CommonCapabilityInquiredType.FIXED_VALUE, UpscalingEffectType.DSEE_HX, UpscalingEffectStatus.INVALID)
⬇ 1 EQEBB_RET_EXTENDED_INFO(EqEbbInquiredType.PRESET_EQ, [(EqBandInformationType.SPECIFIC_INFORMATION, 0x00 0x01), (EqBandInformationType.SPECIFIC_INFORMATION, 0x01 0x90), (EqBandInformationType.SPECIFIC_INFORMATION, 0x03 0xe8), (EqBandInformationType.SPECIFIC_INFORMATION, 0x09 0xc4), (EqBandInformationType.SPECIFIC_INFORMATION, 0x18 0x9c), (EqBandInformationType.SPECIFIC_INFORMATION, 0x3e 0x80)]

⬆ 0 EQEBB_SET_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.VOCAL, [])
⬇ 0 EQEBB_NTFY_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.VOCAL, [0x0a, 0x10, 0x0e, 0x0c, 0x0d, 0x09]
⬆ 1 EQEBB_GET_EXTENDED_INFO(EqEbbInquiredType.PRESET_EQ)
⬇ 1 COMMON_NTFY_UPSCALING_EFFECT(CommonCapabilityInquiredType.FIXED_VALUE, UpscalingEffectType.DSEE_HX, UpscalingEffectStatus.INVALID)
⬇ 0 EQEBB_RET_EXTENDED_INFO(EqEbbInquiredType.PRESET_EQ, [(EqBandInformationType.SPECIFIC_INFORMATION, 0x00 0x01), (EqBandInformationType.SPECIFIC_INFORMATION, 0x01 0x90), (EqBandInformationType.SPECIFIC_INFORMATION, 0x03 0xe8), (EqBandInformationType.SPECIFIC_INFORMATION, 0x09 0xc4), (EqBandInformationType.SPECIFIC_INFORMATION, 0x18 0x9c), (EqBandInformationType.SPECIFIC_INFORMATION, 0x3e 0x80)]

⬆ 0 EQEBB_SET_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.TREBLE, [])
⬇ 1 EQEBB_NTFY_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.TREBLE, [0x0a, 0x0a, 0x0a, 0x0c, 0x10, 0x14]
⬆ 1 EQEBB_GET_EXTENDED_INFO(EqEbbInquiredType.PRESET_EQ)
[... cut ...]
⬇ 0 COMMON_NTFY_UPSCALING_EFFECT(CommonCapabilityInquiredType.FIXED_VALUE, UpscalingEffectType.DSEE_HX, UpscalingEffectStatus.VALID)
⬇ 1 EQEBB_RET_EXTENDED_INFO(EqEbbInquiredType.PRESET_EQ, [(EqBandInformationType.SPECIFIC_INFORMATION, 0x00 0x01), (EqBandInformationType.SPECIFIC_INFORMATION, 0x01 0x90), (EqBandInformationType.SPECIFIC_INFORMATION, 0x03 0xe8), (EqBandInformationType.SPECIFIC_INFORMATION, 0x09 0xc4), (EqBandInformationType.SPECIFIC_INFORMATION, 0x18 0x9c), (EqBandInformationType.SPECIFIC_INFORMATION, 0x3e 0x80)]
```

denisbarac, 2022-02-28:
> Nice, we’re getting somewhere! :D I removed all ACK packets from the data and annotated the rest. The model of the equalizer settings seem to be 6 values between 0 and 20, representing the -10 to [...]

Regarding the equalizer settings, in the Sony Headphones app there are only 5 values. The 6th value is a different setting, called "Clear Bass", presented as a horizontal slider, not vertical. Which explains why you see it separately. However, the default value is still 0 and it still ranges from -10 to 10, so I don't think that should be a problem.

semvis123, 2022-02-28:
> any idea what EBB stands for?

Looking at the apk resources, it looks like Electro Bass Booster, which is also called clear bass.
```xml
    <string name="EBB_Indicator">Electro Bass Booster</string>
    <string name="EBB_Param_Title">Level</string>
    <string name="EBB_Title">CLEAR BASS</string>
```
What is weird is that on the WH-1000XM3, when I look at the live object of `HPCDeviceCapabilityTableset1` it doesn't have an `EBBCapability` instance (maybe EBBCapability is Clear Bass without EQ).

> `⬇ 1 EQEBB_NTFY_PARAM(EqEbbInquiredType.PRESET_EQ, EqPresetId.OFF, [0x0a, 0x0a, 0x0a, 0x0a, 0x0a, 0x0a]`

Looks like the the values are offset by 10, so -10 would be `0x00` (0) and +10 would be `0x14` (20).

This is the capture of changing a single slider (in the manual preset) by 1.
```
2022-02-28 21:43:31.801 MDRPlugin[18499:34961658] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffb9 0x3c 
2022-02-28 21:43:31.862 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-28 21:43:31.869 MDRPlugin[18499:34959394] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffba 0x3c 
2022-02-28 21:43:31.951 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-28 21:43:31.956 MDRPlugin[18499:34961658] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffb9 0x3c 
2022-02-28 21:43:32.011 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-28 21:43:32.014 MDRPlugin[18499:34959394] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffba 0x3c 
2022-02-28 21:43:32.072 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-28 21:43:32.078 MDRPlugin[18499:34961658] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffb9 0x3c 
2022-02-28 21:43:32.116 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-28 21:43:32.119 MDRPlugin[18499:34959394] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffba 0x3c 
2022-02-28 21:43:32.152 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-28 21:43:32.155 MDRPlugin[18499:34961658] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffb9 0x3c 
2022-02-28 21:43:32.165 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-28 21:43:32.167 MDRPlugin[18499:34959394] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffba 0x3c 
2022-02-28 21:43:32.178 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-28 21:43:32.198 MDRPlugin[18499:34961658] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffb9 0x3c 
2022-02-28 21:43:32.209 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-28 21:43:32.229 MDRPlugin[18499:34959394] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffba 0x3c 
2022-02-28 21:43:32.239 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-28 21:43:32.255 MDRPlugin[18499:34961658] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffb9 0x3c 
2022-02-28 21:43:32.265 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-28 21:43:32.272 MDRPlugin[18499:34959394] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffba 0x3c 
2022-02-28 21:43:32.282 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
2022-02-28 21:43:32.298 MDRPlugin[18499:34961658] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x00 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffb9 0x3c 
2022-02-28 21:43:32.309 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬇ 0x3e 0x01 0x01 0x00 0x00 0x00 0x00 0x02 0x3c 
2022-02-28 21:43:32.314 MDRPlugin[18499:34961658] [EALogger] jp.co.sony.songpal.mdr.link | 19 ⬆ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x0a 0x58 0x01 0xffffffff 0x06 0x12 0x0a 0x0a 0x0a 0x0a 0x0b 0xffffffba 0x3c 
[... cut ...]
2022-02-28 21:43:34.809 MDRPlugin[18499:34955225] [EALogger] jp.co.sony.songpal.mdr.link | 31 ⬇ 0x3e 0x0c 0x01 0x00 0x00 0x00 0x15 0x5b 0x01 0x06 0x10 0x00 0x01 0x01 0x01 0xffffff90 0x01 0x03 0xffffffe8 0x01 0x09 0xffffffc4 0x01 0x18 0xffffff9c 0x01 0x3d 0x2e 0xffffff80 0x55 0x3c 
2022-02-28 21:43:34.810 MDRPlugin[18499:34961658] [EALogger] jp.co.sony.songpal.mdr.link | 9 ⬆ 0x3e 0x01 0x00 0x00 0x00 0x00 0x00 0x01 0x3c 
```

It seems to send the values very often (on slider move maybe, instead of on value change), but I guess you can just send it once.