Plutoberth/SonyHeadphonesClient #65: Add support for WF-1000XM4 TWS

1oh1, 2022-01-19:
**Headset:** WF-1000XM4
**Client version:** v1.2
**OS:** Windows 11
**Description:** Can't connect
**Errors:** Couldn't connect: 10049

Any change TWS earbuds like the WF-1000XM4 can be supported?

1oh1, 2022-01-19:
I must mention that the touch controls on the earbuds can already be used to turn ANC on/off but ambient sound levels can't be changed

Also, if the function of the touch controls were changed to control volume instead in the Sony Headphones app, then ANC can't be turned on/off with touch controls.

Kjelstad37, 2022-06-10:
Headset: WF-1000XM4
Client version: v1.3
OS: Windows 10
Description: Can't connect
Errors: Couldn't connect: 10049
[image]

drfeelgoodesq, 2022-06-12:
[image]

Exactly the same issue here, would LOVE this to work on the XM-4's I just bought if possible!

pseudobacon, 2022-06-22:
[image]

Same :(

TheFatTurtle, 2022-07-17:
The issue here seems to be that the device uses a different ServiceClassId. Changing SONY_UUID to "81c2e72a-0591-443e-a1ff-05f988593351" makes the application connect correctly, however you still can't change any values

1oh1, 2022-07-17:
Has anyone figured out how the 'Speak to Chat' setting is toggled? That's the most annoying feature that you can't turn off when using the earbuds with a PC and requires connection to the phone to toggle

I wonder if Windows Subsystem for Android can be used to get the Sony Headphones app running on Windows and change the settings on the earbuds? I'll try it out sometime.

1oh1, 2022-07-17:
If this helps someone, I was able to capture Bluetooth logs from my Android phone (Poco F1) while it was connected to WF-1000XM4. The **btsnoop_hci.log** file can be opened in Wireshark.

Bluetooth capture:
[btsnoop_hci.log](https://github.com/Plutoberth/SonyHeadphonesClient/files/9127940/btsnoop_hci.log)

I only toggled the '**Speak to chat**' feature on and off several times. Didn't touch any other settings. I can capture more logs later on while interacting with all the other toggles if that helps.

1oh1, 2022-07-17:
I captured some more logs, but this time I played with almost every feature available in the Sony Headphones app while recording the screen so hopefully that helps match a particular action in the recording with what's observed in the Bluetooth traffic capture.

The screen capture was started at **2022-07-17-21-38-46-724** (India time UTC+05:30)

Bluetooth capture:
[btsnoop_hci.log](https://github.com/Plutoberth/SonyHeadphonesClient/files/9128091/btsnoop_hci.log)

Screen capture:
[Screenrecorder-2022-07-17-21-38-46-724.mp4](https://www.dropbox.com/s/82qsx4tm548t2nn/Screenrecorder-2022-07-17-21-38-46-724.mp4?dl=0)

TheFatTurtle, 2022-07-25:
After looking at more captures I found multiple Service Ids that all allow a connection. 
`inline constexpr const char * UUIDS[] = {
								"81c2e72a-0591-443e-a1ff-05f988593351", 
								"764cbf0d-bbcb-438f-a8bb-6b92759d6053",
								"f8d1fbe4-7966-4334-8024-ff96c9330e15",
								"956c7b26-d49a-4ba8-b03f-b17d393cb6e2",
								"df21fe2c-2515-4fdb-8886-f12c4d67927c",
								"931c7e8a-540f-4686-b798-e8df0a2ad9f7",
								};`

UUID 0 causes headphone chrash on connect. Music stops fails to send data.  
UUID 1 works, no discconnect 
UUID 2 gets stuck sending data. 
UUID 3 works no dissconnect
UUID 4 gets stuck while sending, stuck while disconnect  
UUID 5 gets stuck while sending, stuck while disconnect

crab2313, 2022-07-28:
> After looking at more captures I found multiple Service Ids that all allow a connection. `inline constexpr const char * UUIDS[] = { "81c2e72a-0591-443e-a1ff-05f988593351", [...]

The right UUID is "956c7b26-d49a-4ba8-b03f-b17d393cb6e2". The wire format is similar to WF1000-XM3，but the fields have different meaning. Thanks to snoop log from @1oh1, I found the packet to turn on/off speak to chat feature and verified on my WF1000XM4.

3e0c0100000004f80200010c3c -----------> on 
3e0c0000000004f80201010c3c ------------> off

HiddenWalrus, 2022-09-13:
Would love to see this implemented as well! Any way I could help out with providing logs, etc.?

m10x, 2022-10-27:
Gadgetbridge just implemented support for WF-1000XM4 with all its features supported ([the commit](https://codeberg.org/Freeyourgadget/Gadgetbridge/commit/8fdfbfa97c639dc7ad49439c546ce4f265f70e51)). Based on logs and information from [this github issue](https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2925).
Maybe this is helpful.

tasskakajoe2326, 2022-11-26:
Still not working for me as well.

krauthaus, 2024-03-25:
Hey, i got a pair of wh-1000 MX5. I get the same error. Is there any more debug data I can collect and provide for you to make them work? I relly like the project! :D

chrislikescisco, 2024-04-30:
I just had the same error, has this been fixed as of yet?
[image]