Plutoberth/SonyHeadphonesClient #72: No luck on WH-XB900N, macOS

eriksr, 2022-05-22:
Headset: WH-XB900N
Client version: v1.3
OS: macOS Monterey v12.3.1
Description: App doesn't work.

Specifically: Successfully connect headphones to laptop via bluetooth. Launch SonyHeadphonesClient. Click "Connect to Bluetooth Device" (had to grant permission the first time -- did so). Select headphones from list. "Enable Ambient Sound Control" is already ticked, try to move the slider and the app beachballs and does not recover :(

This is on a Macbook Air early 2015.

Plutoberth, 2022-05-22:
Can you try it on Windows or Linux? The Mac version uses a different GUI, so it might be a bit problematic.

semvis123, 2022-05-22:
This is probably the Monterey issue mentioned in #62

semvis123, 2022-05-23:
Could you try the version from #68?(https://github.com/Plutoberth/SonyHeadphonesClient/suites/6621527127/artifacts/249412614)

Plutoberth, 2023-04-26:
Closing due to inactivity

closed 2023-04-26 by Plutoberth