Plutoberth/SonyHeadphonesClient #77: Request: Add support for XM5

Evo-c, 2022-06-22:
Would it be possible to add support for the XM5, at the moment it's unable to connect and this tool is literally the lifeline for using Sony headphones on a PC.

Tatsh, 2022-08-07:
Just got one of these too and I'll look into what's going on if I get some time. I would really just like to be able to shut off Speak-to-chat on PC since it seems to find a way to turn itself back on. I hate that feature.

Plutoberth, 2022-08-07:
Awesome :)
I finally have some time again, so I'll be able to help :)

Evo-c, 2022-08-19:
I've found that if you turn it off using the phone app it does save when using the PC, I also highly recommend turning off the "hands-free" mode as Windows tends to switch to that and not switch back again so you end up using that as your main output which is really bad quality.

To save time for those interested, to disable hands-free open settings, select "Bluetooth & devices" then "Devices" and scroll down and click "More devices and printers settings" then right click properties the XM5 and untick "Hands-free telephony" under services.

Tatsh, 2022-08-19:
Linux is switching back and forth to hands-free thanks in part to Pipewire. It's not perfect but it is working unlike a while ago. I think only once it failed to switch back after ending a call. Chrome can handle the switching, but not sure what the protocol is to trigger it on Linux.

Tatsh, 2022-08-27:
The service UUID on this headset is `956c7b26-d49a-4ba8-b03f-b17d393cb6e2`. Once I switch to that it can get status and send commands but sending commands does not do anything.

Ongoing there needs to be an abstraction layer to separate commands for different models because they don't seem to be the same on every model.

Tatsh, 2022-08-27:
I think the command set being used has changed. In the Android code there's reference to `message.mdr.v2.table1` where as the currently supported commandsare in `mdr.v1`.

There are more values for ASM inquired type:

```
   ASM_ON_OFF((byte)33),
   ASM_SEAMLESS((byte)34),
   MODE_NC_ASM_AUTO_NC_MODE_SWITCH_AND_ASM_SEAMLESS((byte)21),
   NC_AMB_TOGGLE((byte)48),
   NC_MODE_SWITCH_AND_ASM_ON_OFF((byte)18),
   NC_MODE_SWITCH_AND_ASM_SEAMLESS((byte)20),
   NC_ON_OFF((byte)1),
   NC_ON_OFF_AND_ASM_ON_OFF((byte)17),
   NC_ON_OFF_AND_ASM_SEAMLESS((byte)19),
   OUT_OF_RANGE((byte)-1);
```

laurinneff, 2022-11-24:
@Tatsh you can toggle speak to chat directly on the headphones by holding the touchpad for roughly one second with two fingers. This doesn't seem to be documented anywhere but I found out by accident recently.

Tatsh, 2022-11-24:
> @Tatsh you can toggle speak to chat directly on the headphones by holding the touchpad for roughly one second with two fingers. This doesn't seem to be documented anywhere but I found out by accident recently.

I found that out too recently.

Evo-c, 2022-12-14:
> @Tatsh you can toggle speak to chat directly on the headphones by holding the touchpad for roughly one second with two fingers. This doesn't seem to be documented anywhere but I found out by accident recently.

You have no idea how much this has been frustrating me! So good to know, thanks!

macoslav, 2023-03-13:
Hi there! Any updates on XM5 support? It would be a life saver!

EndrII, 2023-03-28:
Hi, what about support WH-1000XM5 ?
I use Ubuntu, I can buy these Headphones or no ?

Tatsh, 2023-03-28:
You would be better off getting XM4 if you want to use this app, unless you want to help in development.

Vascro, 2023-10-06:
Any update on XM5 support?

NorthernLightzz, 2023-11-05:
I have the XM5's if you still need help with development

Fot3k, 2023-11-06:
Also just got the XM5's, would love if they were supported.

PrismaticSpirit, 2023-12-10:
Still waiting for updates on XM5 support. Error code '10049' persists.

Plutoberth, 2023-12-23:
@reformat0r helpfully posted this comment over in #105. I'm copying it over here because #105 is a duplicate.

```
Change this part [here](https://github.com/Plutoberth/SonyHeadphonesClient/blob/master/Client/Constants.h#L11-L15) to:

inline constexpr auto SERVICE_UUID = "956C7B26-D49A-4BA8-B03F-B17D393CB6E2";
inline unsigned char SERVICE_UUID_IN_BYTES[] = { // this is the SERVICE_UUID but in bytes
	0x95, 0x6C, 0x7B, 0x26, 0xD4, 0x9A, 0x4B, 0xA8, 0xB0, 0x3F, 0xB1, 0x7D, 0x39, 0x3C, 0xB6, 0xE2
};

And also [this](https://github.com/Plutoberth/SonyHeadphonesClient/blob/master/Client/IBluetoothConnector.h#L6), accordingly:

inline constexpr auto SONY_UUID = "956C7B26-D49A-4BA8-B03F-B17D393CB6E2";

Will allow you to connect. Didn't test further.
```

...so it looks that at least SOME of the issue is that we just didn't have the correct UUID configured.
I don't have the XM5 so I'm unable to assist here. If you want code review I'll be happy to help :)

nightbread, 2023-12-23:
There's a lot more to it than just the UUID. The interface has changed significantly since XM4.

reformat0r, 2023-12-24:
```
Device AC:80:0A:XX:XX:XX (public)
    Name: WF-1000XM5
    Alias: WF-1000XM5
    Class: 0x00240404 (2360324)
    Icon: audio-headset
    Paired: yes
    Bonded: yes
    Trusted: yes
    Blocked: no
    Connected: yes
    WakeAllowed: yes
    LegacyPairing: no
    UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
    UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
    UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
    UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
    UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
    UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
    UUID: A/V Remote Control Cont.. (0000110f-0000-1000-8000-00805f9b34fb)
    UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
    UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
    UUID: Headset HS                (00001131-0000-1000-8000-00805f9b34fb)
    UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
    UUID: Vendor specific           (443cce33-e85d-4b85-8d53-6e319ede53ae)
    UUID: Vendor specific           (45c93e07-d90d-4b93-a9db-91e5dd734e35)
    UUID: Vendor specific           (81c2e72a-0591-443e-a1ff-05f988593351)
    UUID: Vendor specific           (8901dfa8-5c7e-4d8f-9f0c-c2b70683f5f0)
    UUID: Vendor specific           (931c7e8a-540f-4686-b798-e8df0a2ad9f7)
    UUID: Vendor specific           (956c7b26-d49a-4ba8-b03f-b17d393cb6e2)
    UUID: Vendor specific           (9b26d8c0-a8ed-440b-95b0-c4714a518bcc)
    UUID: Vendor specific           (df21fe2c-2515-4fdb-8886-f12c4d67927c)
    UUID: Vendor specific           (f76acb00-7cab-495f-bb1a-e664598fd77f)
    UUID: Vendor specific           (f7a96061-a1b3-40de-aff0-e78ec45a151e)
    UUID: Vendor specific           (f8d1fbe4-7966-4334-8024-ff96c9330e15)
```

mos9527, 2024-03-03:
With some luck, I've got Ambient Sound Mode & Noise Cancelling settings to work on my WF-1000XM5 on its latest firmware (3.0.1 as of the time of my testing).
The WH model is untested, though it probably should work,too.
Testing is very limited and has only been done on Windows. [The fork](https://github.com/mos9527/SonyHeadphonesClient) and [the changes](https://github.com/mos9527/SonyHeadphonesClient/commit/9c827b5edc3c40cbcf4c2280212c4821421eaa5f) are here if anyone's interested. It's *very* hacky though...