Plutoberth/SonyHeadphonesClient #82: building on aarch64 does not work

Stunkymonkey, 2022-11-19:
I am the package-maintainer of the nix-package. But currently it does not build out of the box.
I am currently getting this issue:
```
building
build flags: -j2 SHELL=/nix/store/99bgmihz5ybclg11h00nm9lgqmsfhp4p-bash-5.1-p16/bin/bash
[ 11%] Building CXX object CMakeFiles/SonyHeadphonesClient.dir/ByteMagic.cpp.o
[ 11%] Building CXX object CMakeFiles/SonyHeadphonesClient.dir/BluetoothWrapper.cpp.o
In file included from /build/source/Client/CommandSerializer.h:2,
                 from /build/source/Client/BluetoothWrapper.h:4,
                 from /build/source/Client/BluetoothWrapper.cpp:1:
/build/source/Client/Constants.h:110:18: error: enumerator value '-1' is outside the range of underlying type 'char'
  110 |  OUT_OF_RANGE = -1
      |                  ^
/build/source/Client/Constants.h:129:18: error: enumerator value '-1' is outside the range of underlying type 'char'
  129 |  OUT_OF_RANGE = -1
      |                  ^
In file included from /build/source/Client/ByteMagic.h:9,
                 from /build/source/Client/ByteMagic.cpp:1:
/build/source/Client/Constants.h:110:18: error: enumerator value '-1' is outside the range of underlying type 'char'
  110 |  OUT_OF_RANGE = -1
      |                  ^
/build/source/Client/Constants.h:129:18: error: enumerator value '-1' is outside the range of underlying type 'char'
  129 |  OUT_OF_RANGE = -1
      |                  ^
make[2]: *** [CMakeFiles/SonyHeadphonesClient.dir/build.make:90: CMakeFiles/SonyHeadphonesClient.dir/ByteMagic.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/SonyHeadphonesClient.dir/build.make:76: CMakeFiles/SonyHeadphonesClient.dir/BluetoothWrapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/SonyHeadphonesClient.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

for version `1.2` I did remove the line containing `UNKNOWN = -1`. For the newer version `OUT_OF_RANGE` appeard as well. So I am suggesting in fixing this bug upstream, then patching it all the times.

I would be happy to have a proper fix. Maybe setting it to the Maximum would result in the same behavior?

Plutoberth, 2022-11-21:
I'm very time constrained currently. Please suggest a fix and I'll be happy to merge. I suspect that the issue is that I specified the enum's type - if you can work out places that rely on that (sizeof?) and fix it there too, then that could work. 

On another note, I didn't know that there's a package, that's really cool 😎

Stunkymonkey, 2022-11-22:
my current fix is patching it to `255` which is the maximum of the `char`. Not sure if that is a good idea... maybe @andro2157 can say somthing about it? Since he is the one introducing the `OUT_OF_RANGE` variable.

andro2157, 2022-11-22:
That's interesting. I copied the enums from the official Sony app apk. The `OUT_OF_RANGE` variable is defined like this :
```java
OUT_OF_RANGE((byte) -1);
```
The `byte` type in java is an equivalent for the `signed char` in C++. I guess that here there's an ambiguity about the use of the `char` type. I used it as an implicit `signed char`, so changing the `char` type to a `signed char` would be a proper fix I guess ?
If the `char` type is interpreted as a `signed char`, I guess that `255` might cause a cast warning on some compilation settings.

Stunkymonkey, 2022-11-22:
so you suggest having something like?:
```
enum class VPT_INQUIRED_TYPE : signed char {
```
Just assuming this here: https://en.wikipedia.org/wiki/C%2B%2B11#Strongly_typed_enumerations
This kind of fix would be easy.

andro2157, 2022-11-22:
Yep exactly!

closed 2022-11-23 by Plutoberth