Plutoberth/SonyHeadphonesClient #85: Enable/Disable Quality Mode?

waltercool, 2022-12-08:
As the title says, any way for to switch quality/stable mode?

WH-1000XM2/XM3 does not support equalizer with LDAC, which is quite sad since you lose Clear Bass at phone.

When you connect it back to PC, you likely don't want to use SBC or similar. 

Sony design for this is quite ****** since they enable/disable Bluetooth codecs from headphones to ensure devices won't use high quality ones in case you require simpler codecs.

Plutoberth, 2022-12-12:
I am sure that it's possible, after all we can do anything the app can do, but I'm not entirely sure how

waltercool, 2022-12-12:
Question, is there any good tip you can give me to debug this with Bluez or similar? Maybe I can dig around and find out (scary)

Ideally any tip how to capture the packet, then play around with it would be helpful.

Plutoberth, 2022-12-12:
haha, it's not scary at all, just takes a bit of time :)

There's some good info in the "Protocol Information" section in the project's readme, I recommend that you look there. If you want to play around with it you can just modify the client. I recommend posting your advances and information here, other developers also read those pages and it's likely that they'll collaborate. Have fun!

waltercool, 2022-12-26:
Hi,

Just made those steps, I don't understand much the logs from Wireshark, but basically:

- Started BT with HCI snoop log: "Enabled Filtered"
- At first seconds, Bluetooth connected to headset using Quality Mode (LDAC).
- Waited almost a minute (logs say second 57) and changed to Stable Mode.
- Waited a minute (logs says second 113) and changed back to Quality Mode, still at SBC because requires manual disconnection.

Attached HCI snoop dump. I know MAC Addresses are published inside, may delete log later.

Used: Xperia phone, Sony WH-1000XM3 headphone.

[btsnoop_hci.log](https://github.com/Plutoberth/SonyHeadphonesClient/files/10304764/btsnoop_hci.log)

jollySleeper, 2023-02-22:
Hi, @Plutoberth thanks for this amazing app. 
I used to find Headphones client for desktop every month and I finally found it today.

BTW, is it possible to implement a power off button like we have it in app.
Just curious, I read this thread and will do the steps necessary if this feature can be implemented.

Thanks Again