Plutoberth/SonyHeadphonesClient #86: Looking for a new maintainer

Plutoberth, 2022-12-20:
Hey everyone,

I started this project a few years ago with a cool goal in mind - imitating the functionality of the smartphone app. Over the time, it grew way beyond my expectations - many people have contributed to the project, and tens of thousands have downloaded it!

However, my life has changed since those days, and I can no longer allocate enough time to maintain this awesome project. 

Therefore, I'm **looking for a new maintainer**.  Maintaining this project could be a really cool opportunity to develop C++ and research skills, and I'll of course be available to assist with any questions.

If you're interested, please comment here with a few words about your experience and a way I could contact you :)

Nir ❤️

Dragonslayersakata, 2023-01-25:
Dude, I just found this project! I know nothing about this program, but is there a way provide support (monetarily mostly; I am learning how to code right now) to you or any other developer that picks this project up?

Plutoberth, 2023-02-16:
> Dude, I just found this project! I know nothing about this program, but is there a way provide support (monetarily mostly; I am learning how to code right now) to you or any other developer that picks this project up?

I'd really appreciate [donations](https://paypal.me/plutoberth) , but I cannot guarantee any features from them. Other than that, I think that the best thing would be if someone technical stepped up. I'm really glad that you liked the project!

Plutoberth, 2023-04-07:
> Hey I don't know if you are still looking for a maintainer, but if this project is not going to be actively developed otherwise then I may be able to help.
> 
> I am fairly active on GitHub, which is what I assume is the biggest part of the effort it would take. I am a software developer, but a couple of things to note: 1) I do front-end development now, so haven't even looked at C++ code for a long time. I also haven't done Android app development for even longer, but I assume if it is just reverse-engineering via bluetooth snooping or `adb` that should be fine. 2) I have only one old Windows machine and my Macs are non-Intel, so developing / testing for Windows may be difficult.
> 
> However, I do have a passion for these headphones, which is why I am interested in doing what I can to help.

Hey @axlemax ! I'm indeed still looking for a maintainer. I don't really have enough time to maintain the project other than responding to issues or very sporadic work sprees (as you can see in the delay in replying to this comment).

The C++ part of the app is important, but not very complex (apart from a few hotspots), so it shouldn't be very difficult. Android app development isn't really needed, I didn't know anything when I started this project and I don't know much now. 

About the testing part... I think that the best thing to do to fix this would be to finish the move to Qt (or some HTML based platform, which might be easier nowadays), which would make everything cross platform with good performance and mostly rid the need to test on different platforms. I invited you as a collaborator. Please feel free to contact me with any questions.

closed 2023-04-07 by Plutoberth