Plutoberth/SonyHeadphonesClient #96: Control Transparency From Command Line

saihgupr, 2023-06-06:
It would be great to be able to control transparency / noise cancellation from terminal or a script.
Is this possible?

Plutoberth, 2023-06-06:
It's possible, but it'd require a lot of new code. You're welcome to issue a PR (please base it on the qt branch) that adds this functionality :)

NicoWeio, 2024-06-09:
I'd also love to have this!
It might make sense to then split off the GUI completely and have something like a _libsonyheadphonesclient_ instead, which not just our GUI, but any program, could then access.
For my personal use case, creating a widget for KDE Plasma, a CLI should suffice, though.
I'll put this matter on my backlog and hopefully come back to it eventually. :)