Plutoberth/SonyHeadphonesClient #101: Connection Error

Mukkod, 2023-07-23:
Ive downloaded the program and attempted to connect to my XM5's and that doesnt connect. ive also tried my HT-CT260H which thinks about connecting for a second and then gives the same error. 

the error is simple in description:

Errors:
Couldn't connect: 10049

I understand that XM5 support may not be a thing and if not id love to see it. the CT260H is a sound bar so that might not actually have any settings to fiddle with.

Az1977, 2023-07-31:
Hi, I also have the same issue - XM5 will not connect.  Is this something that the team are planning?

Plutoberth, 2023-08-05:
Sadly this is not planned as I don't have the XM5s nor the time to develop the feature, and is already documented in #29.

closed 2023-08-05 by Plutoberth