Plutoberth/SonyHeadphonesClient #105: wh-1000xm5

321gone, 2023-12-15:
Headset: Wh-1000-xm5
Client version: v1.3.1
OS: Windows 11
Description: Didn't work, but i didn't expect it to ^^

Figured it wouldn't support the xm5s, they ARE very new, but might as well attempt. They are recognized in device discovery but unable to connect: Error code 10049. Headphones are on software version 2.1.0, but it hadn't worked with the previous versions I tried either.

reformat0r, 2023-12-18:
Change this part [here](https://github.com/Plutoberth/SonyHeadphonesClient/blob/master/Client/Constants.h#L11-L15) to:

```cpp
inline constexpr auto SERVICE_UUID = "956C7B26-D49A-4BA8-B03F-B17D393CB6E2";
inline unsigned char SERVICE_UUID_IN_BYTES[] = { // this is the SERVICE_UUID but in bytes
	0x95, 0x6C, 0x7B, 0x26, 0xD4, 0x9A, 0x4B, 0xA8, 0xB0, 0x3F, 0xB1, 0x7D, 0x39, 0x3C, 0xB6, 0xE2
};
```

And also [this](https://github.com/Plutoberth/SonyHeadphonesClient/blob/master/Client/IBluetoothConnector.h#L6),  accordingly:

```cpp
inline constexpr auto SONY_UUID = "956C7B26-D49A-4BA8-B03F-B17D393CB6E2";
```

Will allow you to connect. Didn't test further.

Plutoberth, 2023-12-23:
Closing as #77 is a duplicate :)

closed 2023-12-23 by Plutoberth

14inchPizza, 2024-09-25:
> Change this part [here](https://github.com/Plutoberth/SonyHeadphonesClient/blob/master/Client/Constants.h#L11-L15) to:
> 
> ```c++
> inline constexpr auto SERVICE_UUID = "956C7B26-D49A-4BA8-B03F-B17D393CB6E2";
> inline unsigned char SERVICE_UUID_IN_BYTES[] = { // this is the SERVICE_UUID but in bytes
> 	0x95, 0x6C, 0x7B, 0x26, 0xD4, 0x9A, 0x4B, 0xA8, 0xB0, 0x3F, 0xB1, 0x7D, 0x39, 0x3C, 0xB6, 0xE2
> };
> ```
> 
> And also [this](https://github.com/Plutoberth/SonyHeadphonesClient/blob/master/Client/IBluetoothConnector.h#L6), accordingly:
> 
> ```c++
> inline constexpr auto SONY_UUID = "956C7B26-D49A-4BA8-B03F-B17D393CB6E2";
> ```
> 
> Will allow you to connect. Didn't test further.
> 
> Edit: See also #77

How do I edit this in the downloaded client. super newb here. TY

Plutoberth, 2024-09-25:
> How do I edit this in the downloaded client. super newb here. TY

There are compilation instructions on the README, it might take you some time but it should be pretty simple.

pasical, 2024-10-13:
I made a fork to adapt to XM5 https://github.com/pasical/SonyHeadphonesClient-For-WH-XM5 and this version has been successful on my XM5. I'm sorry if something goes wrong because I'm also a newbie 🙏

pasical, 2024-10-13:
> I made a fork to adapt to XM5 https://github.com/pasical/SonyHeadphonesClient-For-WH-XM5 and this version has been successful on my XM5. I'm sorry if something goes wrong because I'm also a newbie 🙏

Apparently I didn't notice that https://github.com/Plutoberth/SonyHeadphonesClient/issues/77 had fixed the issue 🥲