Plutoberth/SonyHeadphonesClient #107: WF-C700N Support

traversone, 2023-12-21:
hi all,

when trying to connect this new headset, error 10049 comes out.

May I support the troubleshooting in some way? Thank you!
[image]

Plutoberth, 2023-12-23:
Hi, it's possible that you just need to recompile the app with the correct headset UUID. Please see the comment that was posted by @reformat0r in #105 and report back :)

traversone, 2023-12-23:
unfortunately I feel to be a simple "end user" and I am not able to follow your suggestion.  Anyway, how to find out the headset UUID just to start? Thank you

reformat0r, 2023-12-24:
On Linux, you can use `bluetoothctl`. Issuing the `info`-command yields:

```
Device AC:80:0A:XX:XX:XX (public)
        Name: WF-1000XM5
        Alias: WF-1000XM5
        Class: 0x00240404 (2360324)
        Icon: audio-headset
        Paired: yes
        Bonded: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        WakeAllowed: yes
        LegacyPairing: no
        UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
        [...]
        UUID: Vendor specific           (443cce33-e85d-4b85-8d53-6e319ede53ae)
        UUID: Vendor specific           (45c93e07-d90d-4b93-a9db-91e5dd734e35)
        UUID: Vendor specific           (81c2e72a-0591-443e-a1ff-05f988593351)
        UUID: Vendor specific           (8901dfa8-5c7e-4d8f-9f0c-c2b70683f5f0)
        UUID: Vendor specific           (931c7e8a-540f-4686-b798-e8df0a2ad9f7)
        UUID: Vendor specific           (956c7b26-d49a-4ba8-b03f-b17d393cb6e2)
        UUID: Vendor specific           (9b26d8c0-a8ed-440b-95b0-c4714a518bcc)
        UUID: Vendor specific           (df21fe2c-2515-4fdb-8886-f12c4d67927c)
        UUID: Vendor specific           (f76acb00-7cab-495f-bb1a-e664598fd77f)
        UUID: Vendor specific           (f7a96061-a1b3-40de-aff0-e78ec45a151e)
        UUID: Vendor specific           (f8d1fbe4-7966-4334-8024-ff96c9330e15)
        [...]
```

VilotStar, 2023-12-26:
@reformat0r while that does return the UUID, how do you select the correct one? do you just test all of the UUIDs until you find the one that works?

traversone, 2024-01-05:
I am really sorry but I am not able to perform these checks because of my lack of knowledge on Linux

jonathandw743, 2024-09-10:
I tried this with all the UUIDs what didn't start with `0000` by changing the value in `IBluetoothConnector.h` and it didn't work,

mirh, 2024-09-29:
FWIW full(?) support was just added to GB
https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/4168