Plutoberth/SonyHeadphonesClient #108: Support for WH-CH720

VilotStar, 2023-12-26:
I just get the error code 10049, I have found what I think is the UUID for the device, being either "DFA46D0A-BFF9-71A3-6178-3FC16B0731B6" or "75A623FD-5408-3079-8616-41DE50C773F1" using an app on my phone, I then compiled the app twice to test with these UUIDs, with nothing to work, I cannot use linux bluetoothcli due to me being on windows.

VilotStar, 2023-12-27:
Update: I have found a UUID that seems to connect `00000000-DECA-FADE-DECA-DEAFDECACAFF`, quite a weird UUID out of my list of UUIDs. I do not seem to be able to change any settings however, I'm not sure if this is due to the headphones not supporting the current abilities of the program

progandy, 2024-01-21:
If you meant WH-CH720N, then the UUID should be `956C7B26-D49A-4BA8-B03F-B17D393CB6E2` which is the new sony protocol version. Some functions may work if you simply change the UUID, but there are some protocol differences that may need further implementation changes.

This is probably a duplicate of: https://github.com/Plutoberth/SonyHeadphonesClient/issues/107  
Changing the UUID: https://github.com/Plutoberth/SonyHeadphonesClient/issues/105#issuecomment-1860077269
(Other changes might be necessary to support some protocol differences)

mirh, 2024-09-28:
And the previous one is instead the service number for all the apple-certified accessories.
The first two though, are very much sus and they aren't to be found anywhere else on the internet.