Plutoberth/SonyHeadphonesClient #110: WH-1000-XM5 not working

Kyrostar, 2024-02-15:
Headset: WH-1000-XM5
Client version: 1.3.2
OS: Windows 11 Pro x64
Description: Not working. Just says Couldn't connect: 10049, even though it is literally connected to my PC.
[image]

Kyrostar, 2024-02-15:
[image]
You can see it is connected here

aybruh00, 2024-02-16:
The headphones being connected on windows is different from the app connecting to the headphones. Think of it like your computer being connected to your router vs your computer being connected to the internet. Both are different things but it can connect to the internet only after the computer connects to the router. And even when connected to the router there may not be any internet. 
The application opens a channel to the headphone over bluetooth using something called RFCOMM. Windows shows connected over bluetooth; the app says it could not connect over the bluetooth using RFCOMM. That said, I have no idea why this happens as I only have the WH-1000XM4 (if someone were to gift me an XM5 I would be happy to fix the app :P)