Plutoberth/SonyHeadphonesClient #111: Support Android.

RokeJulianLockhart, 2024-04-25:
I don't know how this is architectured, but since you've managed to support Linux, Windows, and macOS, I expect that you've abstracted everything to matters to a reasonable level, and are using a cross-platform GUI toolkit.

Consequently, I ask.

Plutoberth, 2024-04-27:
This is unplanned until Qt support is finished and merged

RokeJulianLockhart, 2024-04-27:
> https://github.com/Plutoberth/SonyHeadphonesClient/issues/111#issuecomment-2080427595

@Plutoberth, is https://github.com/Plutoberth/SonyHeadphonesClient/issues/56#issue-999713911 the meta tracking issue for that?

ruguysgoingtrickortreating, 2024-08-23:
this would be cool because i wish the official app had a feature where the single/double/triple/quadruple tap functionality would change if only a single earbud was in, with an open source alternative either i could stumble around and maybe get it to work myself or open a feature request here

closed 2025-01-25 by Plutoberth