Plutoberth/SonyHeadphonesClient #113: Change pairing mode sony headphone

mrroooooot, 2024-08-31:
Is there any way to activate pairing mode on Sony headphones using code commands? The touch function on my headphones is broken, and I can't connect to new devices.
Thank you!

aybruh00, 2024-08-31:
which model of headphones do you own? I'm pretty sure my WH-1000XM4s can be done this way, but most headphones have a way to enter pairing modes with the physical buttons. Just need to look in the manuals.

mrroooooot, 2024-08-31:
I use sony WF-1000XM3. My touch button to activate pairing mode is broken. Can you tell me how to use the code to activate this mode? Thank you

aybruh00, 2024-09-01:
This code atm doesn't do anything related to multipoint, so it is of no use to you. My PR does include some features of multipoint but not the new pairing feature. I do plan to add it sometime in the future, but right now I am too busy to do it.
Btw, which devices have you got paired with your earphones right now?

mrroooooot, 2024-09-01:
@aybruh00  I have currently connected it to a desktop computer and an iPhone. I want to activate the connection mode to connect it to a laptop. Can you guide me? I will study it.

aybruh00, 2024-09-01:
If you want to program it yourself, I would suggest cloning my repository, as it has a bunch of additional stuff that would make adding the function easier. 
The problem lies in figuring out what message to send to the headset. For that you would need to somehow see what messages the app sends when you do an action. I am on android, so I can dump my bluetooth snoop logs, but I have no absolutely no idea how to do this on IOS.

mrroooooot, 2024-09-02:
@aybruh00 Thanks for your help. Hope you have time to develop this feature.