Plutoberth/SonyHeadphonesClient #120: Sony WF-1000XM5

Sandrdr, 2025-01-01:
[image]

-- could not connect to bluetooth socket --

Fedora 41, Gnome 47.2, Wayland

dioxias, 2025-01-02:
check out this fork for the modern Bluetooth standard headphones https://github.com/mos9527/SonyHeadphonesClient

Alpakash, 2025-01-06:
Holy shit, thanks @dioxias, actually did fork this repo a couple of years ago and was about to go and develop it. However it seems it got already fixed, nice one. No heavy lifting needed anymore for this, it seems (still need to try your referenced link out).

Plutoberth, 2025-01-25:
That's awesome  - I'm glad that someone took the time to make those changes